ARM-software/tf-issues #394: Reducing boot time: bypassing xtable_create and give ttbr_el# the table address in ROM

Mathieu-Brcm, 2016-04-20:
Hi ATF Team,

mmu setup is called at each BL stages (BL1, BL2, BL31 ...).
In current ATF, correct me if I am wrong but there is no way to bypass xtable_create and call the mmu enable macro with the TTBR address directly.

In addition to arm_configure_mmu_el3 we could have a arm_enable_mmu_el3 that will set the TTBR address (passed in parameter) and enable MMU. This would be called in platform (bl(x)_plat_arch_setup instead of arm_configure_mmu_el3).

This would save a lot of RAM and reduce the boot time.

Is there a better way to do it?
Comments/suggestions?

Thanks,
-Mathieu

danh-arm, 2016-04-21:
Yes, for platforms that can construct the page tables statically (perhaps even in ROM), a significant amount of RAM could be saved by not using the xlat_tables library to construct the page tables at runtime (from a static region mapping). Those static tables might be more difficult to maintain than the static region mappings, however.

Platforms that wish to do this now can simply copy the code from `enable_mmu_elX()`. I agree that it might be useful to provide a variant of these functions taking a translation table base address argument, which doesn't pulll in all the other xlat_tables baggage. A contribution to this effect sounds OK to me.

I don't see a need for `arm_enable_mmu_elX()` functions as these wouldn't provide anything additional to the new `enable_mmu_elX()` variants, and we don't have any strong interest in changing the ARM development platforms in this direction.

Mathieu-Brcm, 2016-04-22:
Thanks Dan,
-Mathieu

closed 2016-04-22 by Mathieu-Brcm