ARM-software/tf-issues #395: Build error for PLAT=zynqmp

sorenb-xlnx, 2016-04-25:
The build for the ZynqMP platform fails on the integration branch, with the following error:
```
plat/arm/common/aarch64/arm_common.c: In function 'plat_get_syscnt_freq':
plat/arm/common/aarch64/arm_common.c:171:40: error: 'ARM_SYS_CNTCTL_BASE' undeclared (first use in this function)
  counter_base_frequency = mmio_read_32(ARM_SYS_CNTCTL_BASE + CNTFID_OFF);
                                        ^
plat/arm/common/aarch64/arm_common.c:171:40: note: each undeclared identifier is reported only once for each function it appears in
```

Bisecting the issue points at:
```
c073fda1c692d7c74415d26fb483d6336330fcc0 is the first bad commit
commit c073fda1c692d7c74415d26fb483d6336330fcc0
Author: Yatharth Kochar <yatharth.kochar@arm.com>
Date:   Thu Apr 14 14:49:37 2016 +0100

    Move `plat_get_syscnt_freq()` to arm_common.c
    
    This patch moves the definition for `plat_get_syscnt_freq()`
    from arm_bl31_setup.c to arm_common.c. This could be useful
    in case a delay timer needs to be installed based on the
    generic timer in other BLs.
    This patch also modifies the return type for this function
    from `uint64_t` to `unsigned long long` within ARM and other
    platform files.
    
    Change-Id: Iccdfa811948e660d4fdcaae60ad1d700e4eda80d

:040000 040000 ec55345914082a7b20c80ab8b6395419b7739d58 575e526098ac381a725a3df5ad3662742cc87cae M      include
:040000 040000 448370c1cb4e3c2065d4466ba7ab419bb4c996a6 a8ed9bd5762a7fe2574cc58493107b0f38b0ee29 M      plat
```
Reverting this commit resolves the issue.

The issue can be reproduced with
```
make PLAT=zynqmp RESET_TO_BL31=1 ERROR_DEPRECATED=1
```

danh-arm, 2016-04-26:
Sorry about that. Fix coming soon.

danh-arm, 2016-04-26:
Note, we've now added a ZyncMP build test in our CI tests to avoid problems like this.

closed 2016-04-26 by danh-arm

sorenb-xlnx, 2016-04-26:
Awesome, thanks a lot! I can confirm that the mentioned PR fixes the issue.