ARM-software/tf-issues #396: Initializing the GICv2 interrupts in BL2?

dirkbehme, 2016-04-28:
By default, the GICv2 interrupts are initialized in BL31:

https://github.com/ARM-software/arm-trusted-firmware/blob/master/plat/arm/common/arm_bl31_setup.c#L201

Now, there is a custom board specific patch which adds the GICv2 interrupt initialization to BL2:

https://github.com/renesas-rcar/meta-renesas/blob/jethro/meta-rcar-gen3/recipes-bsp/arm-trusted-firmware/arm-trusted-firmware/0001-add-bl2-bl31-R-Car-support.patch#L1926

The first, general question is if this is architectural clean? Does it matter whether the GIC initialization is done in BL2 or BL32 from ARM trusted firmware architecture point of view?

If I understand the custom board code above correctly, this is done to initialize a watchdog (interrupt) early in the boot process. Afterwards, all remaining interrupts are initialized in BL31, still. This results in an assert in BL31, because the the PPI initialization is supposed to run only once based on the reset values

https://github.com/ARM-software/arm-trusted-firmware/blob/master/drivers/arm/gic/arm_gic.c#L211

In release build, the SW "works", though. Even with the assert as asserts are taken only in debug build.

Any opinion how to deal with GIC initialization partly in BL2 and partly in BL31? And how to avoid the double initialization, then?

danh-arm, 2016-04-28:
GIC initialization is platform specific; it can be done at whatever point is appropriate for the platform (e.g. BL2, BL31 or BL32). ARM standard platforms do this in BL31 (as you've observed). Architecturally, it seems reasonable to do limited intialization in BL2 then more a complete re-initialization in BL31. What line exactly do you see the assert in? It's not obvious to me.

By the way, the GIC driver you refer to is deprecated; I recommend migration to the new GICv2 driver (`include/drivers/arm/gicv2.h`). Also, the platform port seems to do a lot of work in `bl2_early_platform_setup()` (with the MMU disabled); perhaps some of it should be moved to `bl2_platform_setup()`.

achingupta, 2016-04-28:
Just to add to what Dan said...

It is possible to initialise an instance of the GICv2 architecture in any BL. The interface is entirely memory mapped. Only the security state of a register access is used to distinguish between levels of privilege. 

GIcv3 onwards, within a security state, architectural exception levels are used to determine the level of privilege. There are system registers that are only accessible from EL3. Some of these registers needs to be accessed in order to initialise the GIC correctly e.g. `ICC_SRE_EL3` unless you are in a legacy mode.

Keeping in mind the above, if you move to a platform with a GICv3 instance in it then you might need help from software in EL3 i.e. BL1 to ensure correct GIC behaviour in BL2. 

IMO, from an overall GIC architecture perspective, it is advisable to perform initialisation in software at the highest EL. I think this is mandatory for GICv3 but not for GICv2.

dirkbehme, 2016-04-28:
Ok, many thanks to both of you! I think this question can be closed then.

Just to answer the remaining question

> What line exactly do you see the assert in? It's not obvious to me.

The assert comes from gicd_set_ipriorityr()

https://github.com/ARM-software/arm-trusted-firmware/blob/master/drivers/arm/gic/gic_v2.c#L271

being called a second time from BL31 while it already has run in BL2. Having the initialization already done in BL2, the assumption

/* We have an SGI or a PPI. They are Group0 at reset */

is not met any more calling it a second time because then, they don't have their reset values any more.

danh-arm, 2016-04-28:
Hmm, it sounds like this could be simply fixed by clearing the group setting for the interrupt before setting the priority, like we already do in `arm_gic_distif_setup()` for the SPIs. It appears the new GICv2 driver also has the problem. We can address this as a bug.

Note, the platform port could avoid this problem by just doing the complete GIC setup in BL2 and removing the BL31 code. I'm not sure if that's acceptable or not.

danh-arm, 2016-05-11:
Internal ref: http://jira.arm.com/browse/GENFW-1465

closed 2016-07-18 by danh-arm