ARM-software/tf-issues #397: Question: Core awake in a system with shared power domain

jayannad, 2016-05-02:
Hi,

I have a question related to gicv3_cpuif_enable. The current implementation of gicv3_cpuif_enable mark the core as awake before enabling the CPU interface. How would this work in a system where the power domain is shared by multiple CPUs and each CPU has a separate CPU interface block per CPU? The CA and PS bits in GICR_WAKER would have been cleared when the first core in the domain is powered on. When CPU on finish is called for the second one, it would cause ASSERT from the following function.

void gicv3_rdistif_mark_core_awake(uintptr_t gicr_base)
{
        /*
         * The WAKER_PS_BIT should be changed to 0
         * only when WAKER_CA_BIT is 1.
         */
        assert(gicr_read_waker(gicr_base) & WAKER_CA_BIT);
...
}

Thanks,
Jayanthi

soby-mathew, 2016-05-03:
As I understand, each CPU will have a separate GIC CPU interface and an associated Re-Distributor interface. In the scenario you have described,  GICR_WAKER of first CPU is different from the second CPU. Hence I wouldn't expect any problem. Note that the `gicr_base` parameter to this function points to the CPU specific Re-distributor base and not the global re-distributor frame address.

jayannad, 2016-05-04:
I understand that each CPU has its own GIC Redistributor region. But, when the CPUs are sharing the same power domain, GICR_WAKER of all shared CPUs would indicate the same status. When CPU0 is powered on, all others CPUs using the same power domain are powered on. In that case, there is only one GICR_WAKER status though the copy of GICR_WAKER is available in each CPUs redistributor region.

Isn't better to let the platform decide whether to update the GICR_WAKER per cpu based on Power domain hierarchy before enabling the CPU interface? GICV3 is not aware of the power domain hierarchy. This would work if different interfaces are provided for core awake and enabling CPU interface

athoelke, 2016-05-05:
The GICR_WAKER register is associated with the state of the interaction between the Redistributor and the specific CPU interface. It is not related to the power domain relationships between CPUs. From my reading of the specifications, these registers should not be entangled/shared on a system where multiple CPUs share a power domain.

I think it would be problematic if the GIC changed the value of GICR_WAKER because of another CPU's activity (which is what you are suggesting happens). e.g. when a CPU wakes up and set ProcessorAsleep to 0, then other CPUs have their GICR interface activated as well...

danh-arm, 2016-07-05:
I'm closing this as the question seems to be answered

closed 2016-07-05 by danh-arm