ARM-software/tf-issues #400: Invalid assert from GICv3 driver

jayannad, 2016-05-27:
Hi,
If the platform doesn't use both G0 and G1S interrupts, it is possible that either g0_interrupt_array or  g1s_interrupt_array could be NULL. In that case, the condition to assert from gicv3_distif_init and
gicv3_rdistif_init funtions need to be modified as 
        assert(driver_data->g0_interrupt_array ||
               driver_data->g1s_interrupt_array);
instead of
        assert(driver_data->g1s_interrupt_array);
        assert(driver_data->g0_interrupt_array);

Thanks,
Jayanthi

soby-mathew, 2016-05-27:
Yes, it seems the driver is making the assumption that both G1S and G0 interrupts are populated in `gicv3_rdistif_init()` and `gicv3_distif_init()`. So it not just a matter of modifying the assertions but also modifying the these functions to cater to this scenario (they should skip configuring G1S or G0 if the platform has not specified it).

soby-mathew, 2016-05-27:
Internal ref: http://jira.arm.com/browse/GENFW-1495

closed 2016-09-20 by danh-arm