ARM-software/tf-issues #401: Compiler issue?

jforissier, 2016-06-14:
Current master (https://github.com/ARM-software/arm-trusted-firmware/commit/ae43c9493d2c6a2a878f3cee0c240c69fe88f130) fails to build with the cross-compiler that comes with Ubuntu 16.04:

```
$ export CROSS_COMPILE=aarch64-linux-gnu-
$ ${CROSS_COMPILE}gcc --version | head -1
aarch64-linux-gnu-gcc (Ubuntu/Linaro 5.3.1-14ubuntu2) 5.3.1 20160413
$ make bl1
[...]
bl1/aarch64/bl1_exceptions.S: Assembler messages:
bl1/aarch64/bl1_exceptions.S:53: Error: non-constant expression in ".if" statement
```

But, with an older Linaro cross-compiler:
```
$ export CROSS_COMPILE=/some/path/to/aarch64-linux-gnu-
$ ${CROSS_COMPILE}gcc --version | head -1
aarch64-linux-gnu-gcc (crosstool-NG linaro-1.13.1-4.9-2014.09 - Linaro GCC 4.9-2014.09) 4.9.2 20140904 (prerelease)
$ make bl1
[...]
Built build/fvp/release/bl1.bin successfully
```

How to fix this? Should I report to GCC?

sandrine-bailleux-arm, 2016-06-14:
Hello Jérôme,

Unfortunately, I was not able to get the cross-toolchain you're using. I am running
Ubuntu 14.04 and the latest pre-built cross-toolchain that I can get on Linaro's
website (https://releases.linaro.org/components/toolchain/binaries/) is 5.3-2016.02,
which successfully builds bl1.bin... So I was not able to reproduce the build failure
you reported.

That said, I found an interesting comment in the Linux kernel source code, see:
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/compressed/head.S?id=refs/tags/v4.7-rc3#n1025
The `W()` macro is defined there:
https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/arch/arm/include/asm/unified.h?id=refs/tags/v4.7-rc3#n44
This piece of code looks very similar to the TF `check_vector_size` macro so we could
probably fix the build issue you reported using the same technique.

Looking at the "GNU as" documentation:
https://sourceware.org/binutils/docs/as/ARM-Directives.html
the `.inst.w opcode` assembler directive seems related to the Thumb instruction set.
So I am wondering whether we could fix the issue using another assembler directive
that would explicitly instruct to use the ARM instruction set instead...

Regards,
Sandrine

jforissier, 2016-06-14:
Thanks for your suggestion, but I could not find a solution (yet). I am not that knowledgeable in ARM assembler... I just noticed that, by removing the `.align` directives in `vector_base` and `vector_entry`, the compile error disappears. But that's obviously not an option.
I'll try to find a GCC expert because I'd really like to understand if the bug is with the code or the compiler.

jforissier, 2016-06-16:
FYI, confirmed to be a compiler bug [1]. One workaround is to remove the fill value in the alignment directives (i.e., '.align x' instead of '.align x, 0'). Not sure it is acceptable (I suppose the compiler will inserts nops in this case).

[1] https://lists.linaro.org/pipermail/linaro-toolchain/2016-June/005771.html

sandrine-bailleux-arm, 2016-06-16:
Thanks for the compiler bug report, that's useful information. (I did stumble across
the same assembler error in the past when trying to modify other bits of assembly
code in TF involving .align directives so I'm happy to have a better idea of what
caused it).

The fill value in the alignment directive was actually introduced recently by this
patch of mine:
https://github.com/ARM-software/arm-trusted-firmware/commit/79627dc37259781e578c47e1e63856dd0424b2a2

As explained in the commit message, the idea was to fill the exception vectors with
zero bytes (i.e. illegal instructions) rather than nops, just as an extra protection for an
unlikely scenario... If that's causing build issues then I guess we could revert the patch.

danh-arm, 2016-06-16:
> If that's causing build issues then I guess we could revert the patch
I'd rather not go there unless we start seeing Linaro toolchain releases with this problem. In Jérôme's link, Jim Wilson said "This patch isn't present in the binutils-2.25 that tcwg is using". Hopefully the tcwg won't move up until this is fixed (or they will patch this in their releases).

Thanks Jérôme for reporting and chasing this.

jforissier, 2016-06-16:
Well, I think it's a good idea to pad with illegal instructions. So I think you should keep the code as is (and close this issue). When I rebase my working code onto ARM-TF master, I will either apply the workaround or switch compiler ;)
Thanks!

jforissier, 2016-06-16:
@danh-arm you're welcome. Nice to see we all agree ;)

danh-arm, 2016-07-05:
I'm going to close this. Please re-open if this appears in a Linaro compiler version.

closed 2016-07-05 by danh-arm

paulkocialkowski, 2016-07-13:
FYI I have opened a bug about this issue on the binutils/gas tracker: https://sourceware.org/bugzilla/show_bug.cgi?id=20364

Feel free to contribute to it if you'd like the see this issue solved in future upstream binutils/gas releases!

martinlroth, 2016-07-13:
Sorry, I saw this after submitting the pull request to make the change.  This is blocking updates for the coreboot project, which is already using GAS 2.26.  This was somehow missed when we were updating the toolchains, but unfortunately we can't go back to older toolchains because of other projects.  This issue is not fixed in binutils 2.26.1.

Feel free to handle the pull request however you want:
https://github.com/ARM-software/arm-trusted-firmware/pull/664

We don't use the Linaro toolchain, so the coreboot project is stuck due to this issue right now, and cannot build several platforms due to this issue.

paulkocialkowski, 2016-08-07:
A proper fix for this issue was merged in upstream binutils: https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;h=7ea12e5c3ad54da440c08f32da09534e63e515ca

danh-arm, 2016-08-10:
Note that the next Linaro GCC 6 release will be based on binutils 2.26 but the fix to this issue will be patched into that release:

https://lists.linaro.org/pipermail/linaro-toolchain/2016-August/005854.html
https://lists.linaro.org/pipermail/linaro-toolchain/2016-August/005857.html

Therefore this problem should not be observed in any supported Linaro GCC toolchain versions.

rucoder, 2017-07-19:
Just reproduced it with Linro GCC gcc-linaro-5.4.1-2017.01

aarch64-elf-as --version
GNU assembler (Linaro_Binutils-2017.01) 2.25.2 Linaro 2016_02
Copyright (C) 2014 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `aarch64-elf'.

danh-arm, 2017-07-19:
> Just reproduced it with Linro GCC gcc-linaro-5.4.1-2017.01

Oh dear, that's disappointing. If that's the case then I guess Linaro have stopped patching binutils for some reason. Have you reported this to them, for example on the linaro-toolchain mailing list?

I'm now concerned this problem also exists in [Linaro GCC 6.3-2017.02](https://releases.linaro.org/components/toolchain/binaries/6.3-2017.02/) and [Linaro GCC 6.3-2017.05](https://releases.linaro.org/components/toolchain/binaries/6.3-2017.05/), since these are based on the affected binutils 2.27. [Linaro GCC 7.1-2017.5](https://releases.linaro.org/components/toolchain/binaries/7.1-2017.05/) should be OK as this is based on the fixed binutils 2.28. Let us know if you have any more information here.

Note we haven't seen this ourselves yet since we use the toolchain that is bundled with the [Linaro 17.04 Release](https://community.arm.com/dev-platforms/b/documents/posts/linaro-release-notes-deprecated#LinaroRelease17.04), which is 6.2-2016.11.

vchong, 2017-07-19:
@rucoder @danh-arm If my investigation and asking around are correct:

1. Linaro gcc 4.9 is 'paired' with linaro binutils 2.24. No bug.
2. Linaro gcc 5.x is 'paired' with linaro binutils 2.25.x. Bug.
3. Linaro gcc 6.x is 'paired' with linaro binutils 2.27.x (fix backported). No bug.
4. Linaro gcc 7.x is 'paired' with linaro binutils 2.28.x. No bug.

@rucoder Can you try linaro gcc 6? It should work ok. Please report back if not.

vchong, 2017-07-19:
Upon further research, it looks like not all linaro gcc 5.x are affected as well? At least 5.3-2016.02 and 5.3-2016.05 seem to be ok. Will look into this a bit more and come back.

rucoder, 2017-07-19:
I've tried Linaro 6.1.1 from here (just to make sure I'm using the correct path) https://releases.linaro.org/components/toolchain/binaries/6.1-2016.08/aarch64-elf/

I can confirm that 6.1.1 doesn't have this problem

vchong, 2017-07-21:
It looks like the 'bad' patch was backported, albeit in a slightly different form [1], into the 2.25 branch as well :( and got picked up by Linaro in its 5.4 17.01 and 17.05 releases. Previous 5.x releases should be unaffected. The toolchain team is looking into it.

[1] https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;h=4e4d08cf7399b60668ebfeb1eee8ec

vchong, 2017-11-22:
Fixed in 5.5 17.10 release - http://releases.linaro.org/components/toolchain/binaries/5.5-2017.10.