ARM-software/tf-issues #402: Bakery lock assert seen while onlining a CPU

vwadekar, 2016-06-20:
On our Tegra platform recently we have been seeing the following assert during our CPU stress test.

ASSERT: bakery_lock_release <225> : bakery_ticket_number(my_bakery_info->lock_data)

The test just keeps on offlining and onlining CPUs in a loop. The assert is seen for any of the secondary CPUs and is not bound to  a certain CPU.

Interestingly, if I add "(void)bakery_ticket_number(my_bakery_info->lock_data" in bakery_get_ticket(), then I dont see this issue.

`	++my_ticket;
	my_bakery_info->lock_data = make_bakery_data(CHOSEN_TICKET, my_ticket);
        (void)bakery_ticket_number(my_bakery_info->lock_data;

	write_cache_op(my_bakery_info, is_cached);`

Any help would be appreciated.

soby-mathew, 2016-06-21:
@vwadekar,
Could you give more information regarding the issue? 

1. When you say "recently", do you imply that the normal memory bakery locks have started showing issue after some recent commits, or is it the case the issue could have been present from the beginning and that you have only detected it recently? 

2. At the point where `bakery_lock_release()` is invoked, the cache is OFF during power down and the cache is ON during power up. When the assert() fires, do you see a pattern that the issue always happens when the CPU is powering down or powering up?

vwadekar, 2016-06-21:
1. The issue could have been present from the start. We have started seeing
issues recently.

2. We see the assert always when the CPU is powering up.

soby-mathew, 2016-06-21:
Hi Varun,
I could be wrong here, but in a thought experiment, assuming that when Cache coherency interconnect (CCI) is disabled the cache maintanence operations are not propagated between clusters, I could see a potential issue in the  [assert](https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/locks/bakery/bakery_lock_normal.c#L225) during power-up.

There is a chance that `my_bakery_info` could be stale in the cache of the other cluster. The lock is acquired with caches disabled on the current cluster and, since the cache maintenance operations are not propagated, the stale cache line will not be invalidated. Thus reading it without proper cache maintenance for assert() after caches are enabled  could be problematic. Could you try inserting `read_cache_op(my_bakery_info, is_cached)` before the assert at Line 225 and let me know if there is any improvement?

i.e 
```
my_bakery_info = get_bakery_info(plat_my_core_pos(), lock);
read_cache_op(my_bakery_info, is_cached);
assert(bakery_ticket_number(my_bakery_info->lock_data));
```

vwadekar, 2016-06-21:
Thanks Soby.

I am trying out your suggestion now.

During some internal discussions, some folks here feel that we are missing
a `dsbish` after `dccivac` (for the cache maintenance operation to go
through) from read_cache_op() for the cache enabled case.

I am trying to dig through the ARM ARM to find out if this requirement is
specified there. But if you know this off the top of your head, please let
me know.

FWIW, adding this `dsbish` did not help. I still see the assert.

vwadekar, 2016-06-21:
Hi Soby,

I see that Section "D3.4 Cache Support (Pg D3-1706)" in the ARMv8 ARM that
says this:

"All data cache instructions, other than DC ZVA, that specify an address,
can execute in any order relative to loads or stores that access any
address with the Device memory attribute,
or with Normal memory with Inner Non-cacheable attribute unless a DMB or
DSB is executed between the instructions." So it seems we need a "dsb" there

athoelke, 2016-06-21:
The Trusted Firmware code is assuming the following behaviour (subsection D3.4.8 in ARMv8 ARM rev g):

----------------------

**Ordering and completion of data and instruction cache instructions**

All data cache instructions, other than `DC ZVA`, that specify an address:
* Execute in program order relative to loads or stores that access an address in Normal memory with either Inner Write Through or Inner Write Back attributes within the same cache line of minimum size, as indicated by `CTR_EL0.DMinLine`.

-----------------------

As the address of the cache maintenance and the load/store in the bakery lock code is the same then the CPU implementation must guarantee ordering of these operations.

vwadekar, 2016-06-21:
Thanks Andrew.

This might be true for all CPUs other than Denver. At least on Denver, the
data cache maintenance instructions wont take effect until the dsb/dmb is
issued. Section D3.4.8 narrows this down to only Denver CPUs it seems.

vwadekar, 2016-06-22:
One more reason, we think the "dsb ish" is necessary to order the actual
data read after the data cache invalidate. We believe that the read of the
actual lock data can still happen earlier than the cache maintenance due to
out-of-order execution, causing problems.

athoelke, 2016-06-22:
When I talked to Soby earlier I wondered if reordering might be a problem - specifically if the write and cache operation at the end of `bakery_get_ticket()` was not in program order then we can end up with incoherent lock state that could trigger the assertion. The current implementation relies on the rule I quoted above.
 
So Denver doesn't quite meet the ordering requirement for same-address cache maintenance/load/store operations? The spec on non-matching addresses (your original extract) suggests that a `dmb` operation should be adequate to ensure ordering rather than the heavier weight `dsb`? - does Denver require the full `dsb` treatment?

In that case you would need alternate versions of the `read_cache_op()` and `write_cache_op()` macros in `bakery_lock_normal.c`. `write_cache-op()` will need the dmb/dsb inserted before the cache maintenance operation and `read_cache_op()` will need it inserted after the cache maintenance operation. Might be worth a test?

vwadekar, 2016-06-22:
Thanks Andrew. Just to keep things simple, we tried our test only with A57
CPUs and were still able to see the assert. With the change Soby suggested
earlier, we have not been able to hit the assert. The test has completed 6
hours and is still running. So things are definitely looking good.

For denver, I think we should be OK with a dmb instead of dsb, but I need
to confirm.

I will try your suggestion and modify the macros. But in the upstream tree
we really want to converge to a solution than works for all, Denver
included.

athoelke, 2016-06-22:
If Soby's suggested change is effective then it points towards his scenario (cache maintenance not being broadcast between clusters when out of coherency/cache disabled) as being responsible. If Denver really was reordering load/store and cache maintenance to same address then I would expect things to still be going wrong. In that case, you shouldn't add the dmb/dsb to the cache_op()s.

vwadekar, 2016-06-22:
We came up with a new theory doe the failure. During CPU powerup the lock
is taken with caches off. This means the value is written directly to the
dram. After the caches are enabled, since we did not invalidate the lock,
the CPU snoops the lock data from other CPU's cache instead of getting it
from dram. This causes the CPU to get stale data and hit the assert.

Makes sense?

athoelke, 2016-06-22:
That is what the `dc ivac` instruction in the `write_cache_op()` is meant to do in the case of `!cached`. So the uncached write in `bakery_get_ticket()` followed by `write_cache_op()` should also cause any cache entry to be expelled, so that the later cached read in `bakery_unlock()` will bring the fresh values in from RAM.

Your testing indicates that either an invalidation prior to the cached read (Soby's fix) or an extra read between the write and invalidate (your initial test) appears to prevent the failure. It would seem that with the unmodified code, sometimes the invalidation at the end of `bakery_get_ticket()` is not fully effective (presumably in the presence of other masters concurrently attempting to read that line into the cache)

vwadekar, 2016-06-22:
Yes. So we are planning to move ahead with Soby's fix (invalidate prior to
the read) in our downstream code. I will push a patch upstream as well. I
would still like to discuss the 'dsb ish' in read_cache_op(). Should I open
a new ticket for that discussion?

soby-mathew, 2016-06-23:
>  I will push a patch upstream as well.
Thanks, that will be quite useful.

>  I would still like to discuss the 'dsb ish' in read_cache_op(). Should I open
a new ticket for that discussion?

As Andrew mentioned earlier, you would still face this issue and other problems if dependant address loads are not ordered with respect cache operations. Also, I think the ARMv8 ARM is clear in this respect and it adds performance overhead for all the compliant hardware.  Hence I dont think it is necessary to add dsbish in read_cache_op.

vwadekar, 2016-06-23:
So would you be fixing all the ordering issues with the bakery locks? We
found some during our internal review. Let me know if you want me to push
our version as well.

soby-mathew, 2016-06-24:
Hi Varun,
If there are other issues found, we would like to review them. You could push a branch and we could have further discussions based on that.

Best Regards
Soby Mathew

vwadekar, 2016-06-28:
@soby-mathew 

Hi Soby,

I have pushed our downstream patches as part of PR https://github.com/ARM-software/arm-trusted-firmware/pull/656

Please let me know your thoughts

alexvanbrunt, 2016-06-28:
@athoelke, @soby-mathew 
This isn't sufficient:
> Execute in program order relative to loads or stores that access an address in Normal memory with either Inner Write Through or Inner Write Back attributes within the same cache line of minimum size, as indicated by CTR_EL0.DMinLine.

The problem is that the DC operations need to be ordered with respect to accesses **not** using "Normal memory with either Inner Write Through or Inner Write Back attributes". To get that ordering, there needs to be a DSB.

athoelke, 2016-06-29:
@alexvanbrunt 

Can you explain which sets of cache maintenance & memory accesses need to be ordered than fall outside of that rule?

* The current BL31 code defines the xlat `MT_MEMORY` type as Inner Write Back
* The warm boot path through `psci_entrypoint()` enables he MMU prior to acquiring any of the bakery locks
* The power off path disables the cache but not the MMU before releasing the bakery locks

So all of the bakery lock operations in the upstream code operate on Normal IWB memory, and so this rule should apply any ensure that the DC maintenance is ordered with respect to the related read or write access.

But maybe we have missed an execution path which is insufficiently synchronised?

Clearly if the memory type being used for the locks is defined differently for a specific platform, or if the MMU is disabled at the point that the lock code is executed, then additional synchronisation barriers are necessary. But for systems where this is not done, this extra synchronisation places additional significant overhead on the lock operations.

Our testing shows that best performance is provided by placing the locks in "coherent" memory - i.e. a dedicated page of memory that is marked as 'device' and so is not cached.

alexvanbrunt, 2016-06-30:
@athoelke 
Per the ARMv8 ARM, when SCTLR_EL3.C == 0:
> All data access to Normal memory from EL3, and all Normal memory accesses to the
EL3 translation tables, are **Non-cacheable** for all levels of data and unified cache.

This means that those accesses are not
>Normal memory with either **Inner Write Through or Inner Write Back** attributes
That implies that only a DSB is sufficient to order DC instructions with respect to accesses with SCTLR_EL3.C == 0.

In the power off path, SCTLR_EL3.is set to 0 before releasing the lock. So, there needs to be DSB instructions in the bakery lock after every DC instruction.

athoelke, 2016-07-01:
OK. That is an interesting (and possible correct) way of reading the ARMv8 ARM.

At those points in the execution (when `SCTLR_EL3.C=0` during lock/unlock operations): `SCTLR_EL3.M=1` so translation is enabled and the memory type attributes and access permissions in the translation tables are in force. Thus the bakery lock memory is Normal, inner sharable, inner write back. The ARM is silent about whether `SCTLR_EL3.C=0` makes such memory no longer subject to that specific ordering rule or not.

So you may well be right *on some CPU implementations* depending on how the interaction of `SCTLR_EL3.C` and that ordering rule have been implemented.

If this is the case then the problematic issue is:
* a missing `dsb` before the `dc ivac` in the `!is_cached` path in `write_cache_op()` [1]

This is equivalent to placing a dsb() in bakery_get_ticket() before the final call to `write_cache_op()` - but obviosuly a better implementation because it keeps the `dsb` with the DC operation code.

There is no need to put a `dsb` after the `dc civac` in the `is_cached` path of `read_cache_op()` - as caching is enabled at that point and there is no DC operation in the `!is_cached` path.

There is already a `dsb` after the DC operation in `write_cache_op()`

In order to validate this theory of why the current code is broken - could you test a patch which makes only the change identified in [1], and not the additional changes in your pull request?

vwadekar, 2016-07-01:
Thanks Andrew. I can try the change you suggested in [1].

Alex, do you agree with Andrew's theory?

alexvanbrunt, 2016-07-05:
> a missing dsb before the dc ivac in the !is_cached path in write_cache_op()

There needs to be a DSB in the is_cached path and a DMB in the !is_cached path. The DSB ensures that the !is_cached path observes the cache maintenance. The DMB ensures sequentially consistent ordering on an ARM processor.

As a rule, a DC instruction can be  (and is on at least one ARMv8 processors) treated NOP without a DSB because the effects of DC instruction are not guaranteed to be observed without the DSB.

>So you may well be right on some CPU implementations depending on how the interaction of SCTLR_EL3.C and that ordering rule have been implemented.

I just started working with the ARM Trusted Firmware. Is it meant to work on just a few CPUs or is it meant to work on all CPUs that implement the ARM architecture similar to how the Linux kernel works on all ARMv8 CPUs? If it is the latter, then software cannot make assumptions that are not in the ARMv8 ARM.

athoelke, 2016-07-05:
> There needs to be a DSB in the is_cached path and a DMB in the !is_cached path. The DSB ensures that the !is_cached path observes the cache maintenance. The DMB ensures sequentially consistent [...]

My comment was only about the necessary addition to the existing write_cache-op() macro. The DC operations for writes already included a DSB. However you are right that only a DMB is needed between the write and the cache maintenance not a DSB. With the addition of the missing DMB in the uncached write case the bakery lock data accesses should look like this:

Cached write:

    str w1, [x0]
    dc cvac, x0
    dsb ish

Cached read:

    dc civac, x0
    ldr w1, [x0]

Uncached write:

    str w1, [x0]
    dmb ish
    dc ivac, x0
    dsh ish

Uncached read:

    ldr w1, [x0]

> I just started working with the ARM Trusted Firmware. Is it meant to work on just a few CPUs or is it meant to work on all CPUs that implement the ARM architecture similar to how the Linux kernel [...]

The intention is that the code should run correctly on CPUs that conform to the architectural specification. My observation was that ARM Trusted Firmware team thought that it had made a fair reading of the ARMv8 ARM and the application of the ordering rule. And it appears to work OK on ARM's implementations and SoCs - so this did not appear to be an assumption.

Your reading of that rule places that approach in doubt, and as the spec is not clear on the point and the AVK does not report a fault (presumably) it would be reasonable to argue that the more relaxed CPU implementation is also correct.

Hence, if you can verify that the addition of the DMB in the uncached write sequence resolves the issue you are seeing, we should get that change included in the upstream bakery lock code.

alexvanbrunt, 2016-07-05:
A few corrections:

Cached read (the DC operation is needed to order the DC operation with an uncached write):
```
dc civac, x0
dsb ish
ldr w1, [x0]
```
Uncached write (no need for DC operations):
````
str w1, [x0]
dmb ish
```
Uncached read (DMB is needed for ordering):
```
dmb isb
ldr w1, [x0]
```

Do you have any tests for this code that we can use? The root cause of the problem we were hitting all the time was the missing cache invalidate before the assert when there wasn't any contention. We don't have a test case that stresses the contention case yet.

athoelke, 2016-07-06:
Hi @alexvanbrunt,

We have run stress tests which take CPUs offline and back online using this code on some of the ARM CPU implementations (on a Juno board). This involves concurrent and rapid acquistion and release of multiple locks on multiple CPUs with the caching status changing while the lock is held.

In my last reply I had forgotten that we had also missed the read_cache_op() before the assert()s which either we haven't stressed tested a debug build, or the implementation of the ordering rule on the CPUs we tested made it work out OK. So aside from the DMB in write_cache_op() we need to ensure that read_cache_op() happens as before the read when asserting the lock state in get_ticket() and unlock().

I can see what your proposed changes do - assuming the cached read is responsible for synchronising the cache/memory (which we only failed to do in the asserts) and allowing the uncached write to leave stale data in the cache (which might be there due to other CPUs reading this data with caching enabled).

However, in order to avoid memory bloat (the reason for the "normal memory" implementation of the locks), we pack each CPU's lock data for all locks into a single cache line. We rely on the fact that for each CPU, only that CPU will ever write to that cache line but other CPUs can read from it. But the placement of the CPU data for two locks in a single cache line means that we cannot defer the DC invalidate operation from the uncached-write to the cached-read. If it is deferred, then for locks A and B, the sequence cached-read-A, uncached-lock-A, uncached-lock-B, cached-unlock-B will result in the final write cleaning a cache line that contains an out-of-date value for A.

If performance matters, then we recommend using the original "coherent memory" implementation of the locks and the use a of a page of non-cacheable RAM for the lock data. The cache operations are never needed, and the strong memory ordering of nGnRnE device memory ensures correctness.

alexvanbrunt, 2016-07-06:
Can you share that test (or any other unit tests)? We can run it on a different architecture so that there is some confidence that it works on something other than A57s and A53s.

Have you compared the performance of Device-nGnRnE with the coherent version to placing barriers in the correct place? If anything other than the locks is using Device-nGnRnE memory, I bet it is slower. CPUs need to be able to prefetch memory to perform well. Device memory prevents prefetching.

If we want to go fast, we should leave the cache on an do a normal spinlock with LDXR and STXR. That way the time to acquire a lock doesn't grow quadratically with the number of cores. I would be happy to help make such a modification.

athoelke, 2016-07-06:
Have a rummage around the code: we have spinlocks and use/recommend them for most things that needs locking.

The challenge with the power management code is that current ARM interconnects are managed from the AP, powering a cluster off demands that the interconnect has been told to stop snooping into the cluster before this occurs, to prevent disaster that means that the cluster has to disable allocation in its caches before that, and we need to control races between cores powering off (last core has to disable cache + snooping) and powering on (first one up has to enabled snooping). We need a lock in TF to manage those races, and the lock and unlock operations sometimes occur when caching is disabled - and so in most systemsl Exclusive operations are not an option. Hence the bakery locks.

If you build Trusted Firmware with 'coherent memory' enabled (I can't remember the build option off hand) then it will allocate just one page of SRAM in the memory map to have device attributes, and only place the bakery locks and a couple of other data structures (that have similar cached/uncached access demands) in that memory. So general performance is unchanged, but bakery lock performance is much better (about 3x if I recall correctly) due to the elimination of cache maintenance and data synchronisation barriers all over the place.

alexvanbrunt, 2016-07-06:
Would you object to us creating a third version that works with caches on? Our interconnect does not have that limitation.

danh-arm, 2016-07-08:
The comments in this issue and #403 seem to have merged a bit. Can I ask that this issue concentrates on fixing the (possible) bug due a missing barrier in the existing bakery lock code?

Any further discussion on optimizing synchronization in the CPU_SUSPEND path should happen in #403.

Alex - regarding your "3rd version", this is going to require some open-heart surgery in the generic PSCI code. We are thinking about similar changes to cope with future ARM products. Although we want the same end result (a more optimal PSCI solution), I think the platform assumptions and requirements are a bit different in your and our cases. We will contact you privately to understand these requirements better (although any public information should still be shared in #403).

vwadekar, 2016-07-19:
@danh-arm 

Hi Dan,

I agree that the comments for two issues have merged. To get this issue fixed, I guess just the initial change that Soby mentioned should suffice.

I am not sure if we have reached on a consensus for the data ordering changes in the bakery lock. So I can keep just the actual fix in my PR and then we can discuss on the data ordering in a separate ticket.

vwadekar, 2016-07-19:
FWIW, since I had a pending AI, I am testing the following patch:

diff --git a/lib/locks/bakery/bakery_lock_normal.c b/lib/locks/bakery/bakery_lock_normal.c
index 2dc80f1..67a0b51 100644
--- a/lib/locks/bakery/bakery_lock_normal.c
+++ b/lib/locks/bakery/bakery_lock_normal.c
@@ -82,13 +82,23 @@ extern void *__PERCPU_BAKERY_LOCK_SIZE__;
 
 #define write_cache_op(addr, cached)   \
                                do {    \
-                                       (cached ? dccvac((uint64_t)addr) :\
-                                               dcivac((uint64_t)addr));\
+                                       if (cached) {   \
+                                               dccvac((uint64_t)addr);\
                                                dsbish();\
+                                       } else {        \
+                                               dmbish();\
+                                       }       \
                                } while (0)
 
-#define read_cache_op(addr, cached)    if (cached) \
-                                           dccivac((uint64_t)addr)
+#define read_cache_op(addr, cached)    \
+                               do {    \
+                                       if (cached) { \
+                                               dccivac((uint64_t)addr);\
+                                               dsbish();\
+                                       } else { \
+                                               dmbish();\
+                                       } \
+                               } while(0)
 
 static unsigned int bakery_get_ticket(bakery_lock_t *lock,
                                                unsigned int me, int is_cached)

danh-arm, 2016-07-20:
Hi Varun

> I am not sure if we have reached on a consensus for the data ordering changes in the bakery lock. So I can keep just the actual fix in my PR and then we can discuss on the data ordering in a separate ticket.

Agreed. If you update https://github.com/ARM-software/arm-trusted-firmware/pull/656 to only contain Soby's suggested change, we will merge that.

I'm not convinced yet that any other changes are required to the barriers in the bakery lock code. I suspect in your platforms, a more optimal solution will be to introduce abstractions that allow use of spin locks instead bakery locks. This would only be for platforms that can guarantee coherency immediately after reset. Anyway, that's for discussion in https://github.com/ARM-software/tf-issues/issues/403.

soby-mathew, 2016-07-20:
> #define write_cache_op(addr, cached) \
do { \
if (cached) { \
dccvac((uint64_t)addr);\
dsbish();\
} else { \
dmbish();\
} \
} while (0)

> #define read_cache_op(addr, cached) \
do { \
if (cached) { \
dccivac((uint64_t)addr);\
dsbish();\
} else { \
dmbish();\
} \
} while(0)

Hi Varun,
As I understood the discussion, the sequence you should be testing is:

```
#define write_cache_op(addr, cached)	\
            do { \
                  if(cached) { \
                        dccvac((uintptr_t)addr); \
                  } else {
                        dmbish();\
                        dcivac((uintptr_t)addr));\
                 }
                 dsbish(); \
            } while (0)
```

````
#define read_cache_op(addr, cached)    if (cached) \
           dccivac((uintptr_t)addr)
```

vwadekar, 2016-07-20:
Hi Soby,

I read Andrew's and Alex's comments again and it seems the code I posted
followed their steps.

Andrew/Alex can you please confirm if my understanding is correct?

closed 2016-12-01 by davidcunado-arm