ARM-software/tf-issues #403: Remove PSCI locks from the CPU_SUSPEND path

vwadekar, 2016-06-29:
The PSCI code takes and releases bakery locks in the CPU_SUSPEND entry/exit paths. This adds to the CPU suspend and resume latencies. These locks should be removed from the CPU suspend/resume path to reduce these latencies.

We experimented by removing these locks on Tegra platforms and did not see any crashes/hangs over a 24 hour CPU suspend stress test run. But I understand that these locks might be required for other platforms, so we would be OK with macros to disable taking/releasing these locks.

achingupta, 2016-06-29:
These locks are in generic code to guarantee architecturally correct behaviour during use of mixed memory attributes and have nothing to do with specific platforms. They are used to (from the top of my mind)

1. prevent a race where the last CPU in a power domain (> CPU level) is powering down and another CPU in the same domain wakes up. This ensures that state coordination is done correctly and platform ports are able to manipulate global state consistently e.g. perform actions like exiting interconnect coherency correctly. 
2. maintain coherency of PSCI state coordination data structures since state coordination runs with the D$ enabled during power down and with D$ disabled during power up

In 1., in the absence of these locks the CPU that has woken up could execute the state coordination and the platform suspend finisher code before the last CPU. It would not bother entering interconnect coherency since it does not think it is the first CPU in the power domain. The last CPU could then execute the state coordination code and the platform suspend start code and disable interconnect coherency. The system will be left in an inconsistent state.

The coherency problems posed by the use of mixed memory attributes in the power up/down code paths can be fixed without locks. The solution would require use of semaphores, barriers, careful cache maintenance and involve significant complexity. 

It is not clear what your test does exactly and how many times does it hit the above corner conditions. Instead of a passing test, it would be better if you could justify removal of these locks by theoretically proving correctness of the code in their absence in the scenarios listed above. In that scenario, the locks could be removed altogether instead of on a per platform basis.

danh-arm, 2016-07-05:
Clearly bakery locks are not the most performant synchronization primitive, although they do guarantee correctness. As Achin says, there is scope for improvement by abstracting the use of locks and/or switching to different types of synchronization primitive on some code paths. Removing the locks altogether is not option, though.

Doing this would significantly increase the code complexity and it's currently not clear (to us) what the potential gains are. As part of the PSCI STATS PR (https://github.com/ARM-software/arm-trusted-firmware/pull/652), we introduced a Performance Measurement Framework (PMF). We intend to use this to identify bottlenecks in key use-cases like CPU_SUSPEND, which will help justify enhancements in this area.

Varun - if you have any data from your measurements (i.e. with and without the locks), that would be useful.

danh-arm, 2017-07-06:
The general request here is not valid. There may be more specific locking optimizations that can be done in future but these should be raised separately.

closed 2017-07-06 by danh-arm