ARM-software/tf-issues #404: Cannot wake up in Aarch32 secondary cores that standby in Aarch32

nguyenvuhung, 2016-07-01:
Hi all

Currently I use Arm Trusted Firmware load U-boot 32bit, then U-boot 32 bit load OS 32bit at Normal world (EL1). If I only run 1 core, my OS run well but when I run multi core, the primary can't wake up secondary cpu in aarch32 state. I know ATF use PSCI in order to wake up secondary cpu but if primary cpu is running in aarch32 state, how to wakeup the secondary cpu in aarch32 state. Does anyone guide me the way configure in order to wake up secondary cpu ?

Thanks
David

soby-mathew, 2016-07-01:
Hi David,
The secondary CPUs will always boot in AArch64 state at EL3. If the primary has booted in AArch32, the `PSCI CPU ON` will program the CPU context of the secondary CPU appropriately so that when it jumps to lower exception level (EL1) , it will run in AArch32 mode  (see `psci_get_ns_ep_info()`).

To help debug the issue, you could check the following:
1. Does your platform boot well with 64 bit Uboot and 64 bit OS? If this is the case, it should work well with 32 bit OS without any trouble.
2. Has the secondary  CPU booted at all? 
If not, PSCI was not successful in powering ON the secondary.  The problem could lie in the platform code within Trusted OS corresponding to your platform.
3. If the secondary has powered ON, check whether it crashed in either EL3 or in EL1/2. In this case the trouble could either be in platform code, or in your 32 bit OS.

Hope that helps.

nguyenvuhung, 2016-07-01:
Hi soby-mathew,
I appreciate your support
- My platform run good with ATF + uboot64 + OS 64 with condition I must implement PSCI in OS 64 bit in order to wake up secondary 
- Currently I want to run ATF with uboot + OS 32 with multicore without implement separate PSCI in OS 32 because I know ATF have implemented PSCI. Can you suggest me some approach way in order to take advantage ATF feature?

Thank you very much
David

soby-mathew, 2016-07-01:
I don't seem to fully understand your problem.

> My platform run good with ATF + uboot64 + OS 64 with condition I must implement PSCI in OS 64 bit in order to wake up secondary

Could you clarify which OS you are referring to?
PSCI needs 2 components to work: The PSCI implementation in ATF and appropriate driver in OS to invoke PSCI. When you say, you implement PSCI in OS 64 bit, which component are you referring to?

> Currently I want to run ATF with uboot + OS 32 with multicore without implement separate PSCI in OS 32 because I know ATF have implemented PSCI. Can you suggest me some approach way in order to take advantage ATF feature?

As I mentioned, PSCI needs 2 components to work. The PSCI APIs would need to be invoked from an appropriate driver in your OS 32. Is OS 32 and OS 64 different or is it the same OS built for AArch32 and AArch64?

Best Regards
Soby Mathew

nguyenvuhung, 2016-07-02:
My main problem: ATF 64 EL3  + Uboot 32 EL1 + OS 32 EL1 :=> can't run secondary cpu
1. Could you clarify which OS you are referring to?
I use QNX OS. In OS 64 bit, I  implement api psci_cpu_on(cpuid, entrypoint, context_id) in order to primary cpu wake up secondary cpu jump to its address entrypoint
2.  Is OS 32 and OS 64 different or is it the same OS built for AArch32 and AArch64?
     Same type of OS but modify for 2 different version support 2 differrent mode: one for aarch32 and one for aarch64.
- With OS 32 when I use other loader ( this loader in aarch32 state and dont implement PSCI)in      order to load OS 32 it operate well with SMP (multicore) but when I use ATF it only operate well with 1 cpu, can't wake up secondary cpu. In this case, OS 32 don't use PSCI in order to wake up secondary cpu
- With OS 64 it run well SMP with both ATF and other loader. in this case, OS 64 must implement a PSCI api such as psci_cpu_on(cpuid, entrypoint, context_id).

3 .Do you know which method to disable PSCI in ATF at EL3 in order to It don't effect to OS 32 at EL1 ?.
Because I see most platform use armv7 don't use PSCI in order to wake up secondary cpu

Thank you for your support
David

soby-mathew, 2016-07-04:
> Do you know which method to disable PSCI in ATF at EL3 in order to It don't effect to OS 32 at EL1

You cant disable PSCI in ARM Trusted Firmware. You would need to prevent your OS32 from making any PSCI SMC calls. 

It seems you need to modify OS 32 to work with PSCI in ATF.

danh-arm, 2016-07-05:
It's hard to figure out what the actual question is here. It sounds like you have 2 competing PSCI implementations, one in the OS and one in TF, and that you only want to use the OS implementation for some reason, not the TF one. Presumably the only reason you are using TF is for the cold boot flow? We are not going to be able to advise you on implementing PSCI (or a non-PSCI secondary CPU boot method) in the OS.

It would be cleaner if both your OS 64 and OS 32 used PSCI SMCs, which are handled by the same AArch64 PSCI implementation in TF. We might be able to advise you if you go in that direction.

danh-arm, 2016-07-05:
Again, this doesn't sound like a TF question; it probably should be directed to the U-Boot community. In general we expect the normal world firmware (U-Boot in this case) to execute at the same EL and register width as the OS. I'm not sure what your use-case is but I guess you want a single U-Boot image to be able to load both AArch64 and AArch32 OS, selectable at runtime? For this I can see 2 possible solutions:

1.   U-Boot has an option for dropping down into AArch32 EL1 before loading the AArch32 OS, or
2.   U-Boot starts the OS at AArch64 EL2. The OS entrypoint immediately drops down into AArch32 EL1 before executing the rest of OS init code.

nguyenvuhung, 2016-07-05:
Hi danh-arm,

As you guess, I want to use u-boot 64 boot load both aarch32 OS and aarch64 OS at runtime
I will try to follow your advise

Thank you for your support
David

closed 2016-09-13 by danh-arm