ARM-software/tf-issues #405: cert_create

marcSTM, 2016-07-04:
hello,
I try to use cert_create, but I don't find documentation.

I'm able to generate all certificates with one command, but I would like to generate only certificates for b31 and only certificates for bl33.

Can you tell me the command to:
- generate only the trusted-key.cert?
- generate only cert for bl31 (bl31-cont.cert and bl31key.crt)?
- the same for bl33?

rgds
Marc

danh-arm, 2016-07-05:
The cert_create tool isn't really designed to be used standalone since its options and usage is highly dependent on the Trusted Board Boot Requirements (TBBR) Chain of Trust (CoT), which is subject to change. For example, to generate the BL31 key and content certs, you need to provide (or generate) the corresponding BL31 private key. To generate this key, you may also need to generate several higher keys in the CoT right up to the Root of Trust Private Key.

These complex interdependencies are abstracted away somewhat if you invoke the tool via the top-level Makefile. If you just specify the BL images you want to build (or use as input) and use the GENERATE_COT = 1 option, it will figure out which certs to generate.

Having said that, it is possible to do what you want directly via cert_create. I just don't have time to figure out the exact runes for you. Try looking in the source files in `tools/cert_create/src/tbbr/` to see all the TBBR cert_create options.

marcSTM, 2016-07-06:
hello,
I had a look in the code.
results below

usual usecase: secure team provide to the customer/other team a secure fip (bl2+bl31+bl32) + priv_ns_world_key and the customer add his bl33...

rgds
Marc

/* create secure/non_secure world keys: "non-trusted-world-key.key trusted-world-key.key  trusted_key.crt" */
../tools/cert_create/cert_create -n --trusted-key-cert trusted_key.crt --key-alg rsa --rot-key ../plat/arm/board/common/rotpk/arm_rotprivk_rsa.pem --trusted-world-key trusted-world-key.key --non-trusted-world-key non-trusted-world-key.key -k

/* create bl2 certificate : "tb_fw.crt" */
../tools/cert_create/cert_create -n --key-alg rsa --rot-key ../plat/arm/board/common/rotpk/arm_rotprivk_rsa.pem --tb-fw  ../bl2.bin --tb-fw-cert tb_fw.crt -p

/* create bl31 certificates with my own trusted-world-key: "soc_fw_content.crt soc_fw_key.crt" */
../tools/cert_create/cert_create -n --key-alg rsa --soc-fw  ../bl31.bin --soc-fw-cert soc_fw_content.crt --soc-fw-key-cert soc_fw_key.crt  --trusted-world-key trusted-world-key.key -p

/* create bl31 certificates with my own trusted-world-key and my own soc-fw-key : "soc_fw_content.crt soc_fw_key.crt" */
../tools/cert_create/cert_create -n --key-alg rsa --soc-fw  ../bl31.bin --soc-fw-cert soc_fw_content.crt --soc-fw-key-cert soc_fw_key.crt -p --soc-fw-key soc-fw-key.key --trusted-world-key trusted-world-key.key -p

/* create bl33 certificates with my own non-trusted-world-key.key:  "nt_fw_key.crt and nt_fw_content.crt"  */
../tools/cert_create/cert_create -n --key-alg rsa --nt-fw ../bl33.bin --nt-fw-cert nt_fw_content.crt --nt-fw-key-cert nt_fw_key.crt --non-trusted-world-key non-trusted-world-key.key -p

/* create bl33 certificates with my own non-trusted-world-key.key and my own nt-fw-key :  "nt_fw_key.crt and nt_fw_content.crt"  */
../tools/cert_create/cert_create -n --key-alg rsa --nt-fw ../bl33.bin --nt-fw-cert nt_fw_content.crt --nt-fw-key-cert nt_fw_key.crt --non-trusted-world-key non-trusted-world-key.key --nt-fw-key nt-fw-key.key -p

danh-arm, 2016-07-07:
Thanks for the info. As the TBBR stabilises and is proven in products, there may be more value in this interface being better documented.

closed 2016-09-13 by danh-arm