ARM-software/tf-issues #407: How to avoid false positive warnings on deprecated APIs?

sorenb-xlnx, 2016-07-06:
Compiling the ATF for the ZynqMP platform generates a couple of compiler warnings:
```
make -rRj 16 RESET_TO_BL31=1 PLAT=zynqmp
...
bl31/bl31_context_mgmt.c: In function 'cm_get_context_by_mpidr':
bl31/bl31_context_mgmt.c:105:2: warning: 'platform_get_core_pos' is deprecated [-Wdeprecated-declarations]
  return cm_get_context_by_index(platform_get_core_pos(mpidr), security_state);
  ^
In file included from bl31/bl31_context_mgmt.c:36:0:
include/plat/common/platform.h:284:14: note: declared here
 unsigned int platform_get_core_pos(unsigned long mpidr) __deprecated;
              ^
bl31/bl31_context_mgmt.c: In function 'cm_set_context_by_mpidr':
bl31/bl31_context_mgmt.c:116:2: warning: 'platform_get_core_pos' is deprecated [-Wdeprecated-declarations]
  cm_set_context_by_index(platform_get_core_pos(mpidr),
  ^
In file included from bl31/bl31_context_mgmt.c:36:0:
include/plat/common/platform.h:284:14: note: declared here
 unsigned int platform_get_core_pos(unsigned long mpidr) __deprecated;
              ^
bl31/bl31_context_mgmt.c: In function 'cm_init_context':
bl31/bl31_context_mgmt.c:131:3: warning: 'platform_get_core_pos' is deprecated [-Wdeprecated-declarations]
   cm_init_context_by_index(platform_get_core_pos(mpidr), ep);
   ^
In file included from bl31/bl31_context_mgmt.c:36:0:
include/plat/common/platform.h:284:14: note: declared here
 unsigned int platform_get_core_pos(unsigned long mpidr) __deprecated;
              ^
```

Those are easily suppressed by specifying ```ERROR_DEPRECATED=1``` on the build command line, but I thought it would be nice if these could be suppressed on default and was wondering if anybody might have a good suggestion for how to do that.

I was thinking about adding
```
ERROR_DEPRECATED := 1
```
to the platform.mk. On the one hand, that would solve the false positives now, but on the other probably break compilation in the future if additional APIs are deprecated.

danh-arm, 2016-07-07:
Hi Soren

These deprecated warnings appear when building any platform, not just ZyncMP. In general all warnings are treated as errors but this specific warning type is ignored because we know its harmless. In a future (non-compatible) TF release, it would be good to remove all these deprecated APIs, which would clean up the build output, but we're not ready to do that yet.

If you don't mind your platform being occasionally broken by APIs suddenly becoming deprecated, then it's OK to add `ERROR_DEPRECATED := 1` to your platform.mk. If you prefer better compatibility (at the expense of these kinds of warnings appearing), then leave as-is.

If these warnings are really bugging you (or causing some real problem), then one further possibility would be to define a new build macro (e.g. WARN_DEPRECATED). This would default to 1 and have the same behaviour as currently. If `!WARN_DEPRECATED` then all the deprecated APIs would not use the `__deprecated` modifier so there would be no warnings in the build output (although the deprecated APIs would still be compiled in). A contribution to that effect would be OK, though I don't think it adds much.

Regards

Dan.

sorenb-xlnx, 2016-07-07:
Thanks. I'm just gonna leave it as is.

closed 2016-07-07 by sorenb-xlnx

danh-arm, 2018-02-22:
After looking at this again, I think the best way to remove these false positive deprecated warnings is to use `#pragma GCC diagnostic ignored "-Wdeprecated-declarations"` in the call from the compatibility function(s), since we know this is valid reason for using the deprecated function. That would ensure that any other calls to deprecated functions are still flagged.