ARM-software/tf-issues #408: Make use of DC ZVA instruction when zeroing memory

danh-arm, 2016-07-20:
When zeroing a range of normal memory, it is much more efficient to use the DC VZA instruction (if the implementation supports it) than store instructions . Trusted Firmware currently doesn't make use of this instruction. The DCZID_EL0 register should be read to determine if DC VZA is supported and the block size to use.

In particular:

*   The `memset()` implementation in `lib/stdlib/mem.c` is often called with zero as the val argument, which results in a loop of STRB instructions.

*   The `zeromem16` implementation in `lib/aarch64/misc_helpers.S` uses STP and STRB instructions. This is used for zeroing out the BSS and coherent RAM sections.

There may be other places.

danh-arm, 2016-07-20:
Hmm, just realised that of course the MMU/data cache is typically not yet enabled when zeromem16 is called so this won't help in that case.

Also, I don't know if there are any significant performance gains to be had in any real use-cases to justify this task.

vwadekar, 2016-07-20:
Hi Dan,

At least on Tegra for cleaning up Video memory, ranging in MBs, we could
use this function.

If course, MMU is disabled now before we clean that memory, as it is NS,
but DC zva would be helpful when we fix that problem.

Thoughts?

danh-arm, 2016-07-20:
Hi Varun

Yes, zeroing secure video memory is a compelling use-case for this. We can perhaps look into this once  #310 is implemented. Incidentally, we have been working on that task but it is proving very complex to get right.

The only other place we currently zero out large memory regions is when there is an authentication failure, but optimising for that use-case is not compelling.

vwadekar, 2016-07-20:
Hi Dan,

Good to know that you are working on #310. Once you are ready, I can
quickly test out the patches on Tegra to see if things are still OK.
Meanwhile, "dc zva" will have to wait, I guess?

danh-arm, 2016-07-21:
> Meanwhile, "dc zva" will have to wait, I guess?

Yes, I was just noting this down as a(nother) potential task. No plans from our side to do it yet.

closed 2017-02-20 by danh-arm

xuqingqing7, 2019-12-18:
@danh-arm @vwadekar

xuqingqing7, 2019-12-18:
@danh-arm @vwadekar

xuqingqing7, 2019-12-18:
@danh-arm @vwadekar 
Hello, I have read your discussion about the performance effect of DC ZVA. 
Recently I have do some work on this. What confused me is that DC ZVA doesn't bring any performance promotion.

xuqingqing7, 2019-12-18:
@danh-arm @vwadekar 
Hello, I have read your discussion about the performance effect of DC ZVA.
Recently I have do some work on this. What confused me is that DC ZVA doesn't bring any performance promotion.
The test platforms include Cortex A72 and Cortex A76.
The test programs include all of SPEC CPU 2006.

xuqingqing7, 2019-12-18:
@danh-arm @vwadekar 
Hello, I have read your discussion about the performance effect of DC ZVA.

- Recently I have do some work on this. What confused me is that DC ZVA doesn't bring any performance promotion.

- The test platforms include Cortex A72 and Cortex A76.

- The test programs include all of SPEC CPU 2006. Among them, DC ZVA occupies about 20% of execution time of certain program named 403.gcc. It doesn't show any difference in the case of using DC ZVA and prohibiting DC ZVA.

Can you discuss with me about this result in detail? Do you have any interest in this result?
**Looking forward to your reply. Thanks very much.**

sandrine-bailleux-arm, 2019-12-18:
Hello @xuqingqing7 

The TF-A project is no longer active on Github, it has now moved to https://www.trustedfirmware.org/

Could you please send your question to the project's mailing list instead? See:
https://lists.trustedfirmware.org/mailman/listinfo/tf-a

Regards,
Sandrine