ARM-software/tf-issues #409: bl31 booting struck up within opteed_synchronous_sp_entry( )

chinnabha, 2016-07-25:
Hi Experts,

I have compiled the ATF with following bl32 compilation options to load the optee_os via ATF opteed. I also initilaized <static uint32_t next_image_type = SECURE;  @bl31/bl31_main.c>. I am seeing optee_init( ) in bl31 booting struck up in opteed_synchronous_sp_entry( ). 

Could someone please help me if I am missing anything here? I have added bl32 address range to bl2 memory map. I tried adding bl32 address range to bl1 and bl31 memory map. That did not help in solving the issue. 

**NEED_BL32=yes**
**SPD=opteed**
**BL32= <optee_os/out/platform_xxx/core/tee.bin>** 

**The log snippet of the final messages before hanging in optee_init:**
NOTICE:  BL31: v1.2(debug):test
NOTICE:  BL31: Built : 19:47:36, Jul 25 2016
INFO:    GICv3 without legacy support detected. ARM GICV3 driver initialized in EL3
INFO:    BL31: Initializing runtime services
INFO:    BL31: Initializing BL32
INFO:    Entering optee_init
INFO:    Entering opteed_synchronous_sp_entry()

Thanks,
Chinnabha

danh-arm, 2016-07-26:
I suspect this is a question for the OPTEE guys. Does the earlier log say that BL2 has loaded BL32 successfully? It sounds like there is either:

1.  A mismatch between the version of OPTEED in TF you are using and the version of OPTEE itself. For example, [this commit](https://github.com/ARM-software/arm-trusted-firmware/commit/29464f13f6f940abf7fb6218df05e38b0bb3f80f) changed OPTEED to assume that it initializes a 64-bit version of OPTEE.

2.  A mismatch between how TF expects OPTEE to have been built. For example, if you switch on paging support in OPTEE then the OPTEE binary will be prefixed by a header that BL2 doesn't know how to parse.

vchong, 2016-07-26:
Please try below patch if it's not in your branch/tree to see if it helps.

https://github.com/linaro-swg/arm-trusted-firmware/commit/f6244e85df1d931e2ed6bd80e76fbf05a1b32f23

chinnabha, 2016-07-26:
Thanks a lot for the help.

Yes, BL32 loaded successfully. Coming to the two suspects:

    1.  I already merged the above commit "opteed_rw = OPTEE_AARCH64" before reporting this issue. The optee_os build cmd that I am using is:
**make PLATFORM=xxxx CFG_ARM64_core=y CFG_TEE_CORE_LOG_LEVEL=4 DEBUG=1** 
Please suggest if I miss any other compilation flag in optee_os compilation above? Is there any chance that optee_os includes its own dispatcher (opteed) integrated into its image for AARCH64?  

    2.  I cross checked the pager enabling. CFG_WITH_PAGER is not enabled in my optee_os.

Hi Victor - I did checkout of ATF and optee_os source code's (from git hub) by end of May 2016. The patch that you suggested seems way back to Nov 2014.

vchong, 2016-07-26:
> make PLATFORM=xxxx CFG_ARM64_core=y CFG_TEE_CORE_LOG_LEVEL=4 DEBUG=1

Just to be thorough, did you specify the cross compilers? Usually, we use:
```
make PLATFORM=xxxx CFG_ARM64_core=y CROSS_COMPILE=xxxx CROSS_COMPILE_core=xxxx CROSS_COMPILE_ta_arm64=xxxx
```

> Is there any chance that optee_os compiles its own dispatcher (opteed) into its image and getting messed up with the opteed in ATF?

No. optee_os can come with its own monitor but only for v7.

> The patch that you suggested seems way back to Nov 2014.

Yes but don't think it has made it to mainline yet. No harm trying to see if it'll work, right?

chinnabha, 2016-07-27:
I am seeing a synchronous interrupt when primary processor is moving from BL31 to BL32 entry point **_start** @optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S

Upon further debug I am seeing 7 words of BL32 image offset from its load address (0x82000000). This looks to be the suspect #2 @danh-arm has mentioned above.

**(gdb) x/100xw 0x82000000**
0x82000000:	0x4554504f	0x00000101	0x000607fc	0x00000000
0x82000010:	0x82000000	0x00084520	0x00000000	**0xaa0003f3
0x82000020:	0x100107e0	0xd518c000	0xd5033fdf	0xd5381000
0x82000030:	0xd2820141	0xaa010000	0xd5181000	0xd5033fdf
0x82000040:	0x14000000	0x940156cb	0xf100041f	0x5401028a
0x82000050:	0xd37df002	0x102b1b81	0xf8626821	0xd50040bf**

Disassembly of section .text:
0000000082000000 <_start>:
**_start():**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:38
    **82000000:   aa0003f3        mov     x19, x0**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:40
    **82000004:   100107e0        adr     x0, 82002100 <reset_vect_table>**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:41
    **82000008:   d518c000        msr     vbar_el1, x0**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:42
    **8200000c:   d5033fdf        isb**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:44
    **82000010:   d5381000        mrs     x0, sctlr_el1**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:45
    **82000014:   d2820141        mov     x1, #0x100a                     // #4106**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:46
    **82000018:   aa010000        orr     x0, x0, x1**
/home/chinnabha/Documents/optee_os-2.0.0/core/arch/arm/kernel/generic_entry_a64.S:47
    **8200001c:   d5181000        msr     sctlr_el1, x0**

@danh-arm , could you please tell me how do I switch off the paging support? 

@vchong, what the xxxx in CROSS_COMPILE_core=xxxx and CROSS_COMPILE_ta_arm64=xxxx refers to? Platform? Please provide me an example.

vchong, 2016-07-27:
xxxx is your cross compiler, i.e. arm-linux-gnueabihf- or aarch64-linux-gnu-.

Can you please just try the patch I gave you? The 7 words is the size of the header that BL2 doesn't know how to parse, and the patch does!

chinnabha, 2016-07-27:
Thanks @vchong for the help. Sure, I will do that. If the patch knows header parsing then it definitely fixes the issue. Do you you have plans to mainline this patch? If there is no plan then it won't help me permanently. Rather I will prefer to figure out the feature contributing to this header. 

Please advise if there is a way to avoid this BL32 image header in FIP. Avoiding header formation seems easy solution for me.

vchong, 2016-07-27:
I know of efforts to mainline the patch but not of what the status is. Maybe related parties can help chip in here.

To avoid the header you'll have to modify optee_os to build without it, which is probably more than just a single patch and not mainline either.

danh-arm, 2016-07-27:
Upstreaming support for OPTEE header parsing is a work in progress but it's not going to happen soon. In the meantime, it's preferable (from a TF perspective) to enable building OPTEE with or without the header, using the latter configuration for now. I don't know how to build OPTEE in this way or what additional work is needed to make this happen (if any).

chinnabha, 2016-07-28:
_"To avoid the header you'll have to modify optee_os to build without it, which is probably more than just a single patch and not mainline either."_
@vchong, please help me providing the list of optee_os patch numbers by which I can avoid its header compilation? ATF patch, linaro-swg/arm-trusted-firmware@f6244e8 requires changes to /include/plat/xxxx/common/xxxx_def.h, and I am not comfortable making changes to this file. The patch assumes SRAM availability which is not true in my case. 

FIP should pack BL32 and BL33 in the same way. There is no issue with BL33 loading. Does this imply BL33 compilation has no header? Or BL2 has the mechanism to remove BL33 header?

vchong, 2016-07-28:
@chinnabha, [1] is the original patch that added support for the header. There's not an option to build without a header atm, and not sure if there's an easy or straightforward way to undo the process by just avoiding a list of patches. Since you're not using paging support, there's a chance it might work by just removing the header (first 7 words) from tee.bin before FIP packing.

[1] https://github.com/OP-TEE/optee_os/commit/8f7de3fc92dbdfdc217b83403c64abc0bafb5d31#diff-c129fd4d0e6042984a6fcca0603e63d6

> FIP should pack BL32 and BL33 in the same way. There is no issue with BL33 loading. Does this imply BL33 compilation has no header? Or BL2 has the mechanism to remove BL33 header?

The header is an OP-TEE-specific implementation for paging support. It has nothing to do with FIP packing, BL33, BL2 or even other BL32 images.

vchong, 2016-07-28:
Btw, the patch doesn't assume SRAM availability.

chinnabha, 2016-08-01:
Yes, removing the 7 words header from tee.bin solving my issue. Thanks for the help @vchong

closed 2016-08-01 by chinnabha

vchong, 2016-08-16:
@chinnabha FYI, just found out that you can just use `tee-pager.bin` in `optee_os/out/arm*/core/` directly, which is the 'header-less' version. Hopefully can save you some time/work not having to do the removal manually.

chinnabha, 2016-09-12:
Thanks much @vchong. Somehow I missed this message.