ARM-software/tf-issues #411: GICv3 Driver issue

raghuncstate, 2016-08-03:
gicd_write_ipriorityr(gic_common.c) and gicr_write_ipriorityr(gicv3_helpers.c) are written in such a way as to write the priority of 4 interrupts at a time. However, functions gicv3_secure_spis_configure() and gicv3_secure_ppi_sgi_configure() in gicv3_helpers.c operate on an interrupt by interrupt basis. These functions use the aforementioned ipriority write functions on an interrupt by interrupt basis and hence, writing the priority of one interrupt, clobbers the priority of the 2 other interrupts in the same gicd/r_ipriority register. EIther the write functions or the *configure functions need to be fixed.

soby-mathew, 2016-08-04:
> However, functions gicv3_secure_spis_configure() and gicv3_secure_ppi_sgi_configure() in gicv3_helpers.c operate on an interrupt by interrupt basis. These functions use the aforementioned ipriority [...]

The `gicv3_secure_spis_configure()` and `gicv3_secure_ppi_sgi_configure()` use the `gicd_set_ipriorityr()` accessor to write the priority of an interrupt ID.  Previously it was using the `gicd_write_ipriorityr()` which was corrected by the patch here :

https://github.com/ARM-software/arm-trusted-firmware/commit/38a7861450409b6b234e12f15b8b516aa71b6610

danh-arm, 2016-08-18:
Closing this as the issue has already been addressed as Soby describes

closed 2016-08-18 by danh-arm