ARM-software/tf-issues #415: self testing of trusted firmware possible?

monkey-jsun, 2016-08-07:
Hi, all,

I'm new to TF.  So this might be a very dumb question.

I see all the examples are given to run TF with kernels or UEFI.  I wonder if there is a simple testing payload that can be loaded just to verify sw and hw are in a sane state.

I saw something like "test secure payload".  Is that a test suite suitable for this purpose?  The instruction is clear on how to build it, however, it is not clear how to use it with TF.  BTW, I'm currently experimenting with FVP foundation platform.

Jun

danh-arm, 2016-08-10:
Hi Jun

The Test Secure-EL1 Payload (TSP) is more like example reference code than test code, although it does test world switching behaviour in a running system by periodically interrupting the normal world.

I think what you're looking for is a TF bare metal test suite that mostly executes in the normal world, possibly accompanied with an SPD/SP that drives TF functionality not available to the normal world. Such a thing is not available publically but is something we've thought about. There are number of possible requirements here and I'd be interested in any feedback on what would be most useful. For example,

1.  A simple "hello world" payload that confirms the boot sequence has executed through to the normal world.
2.  A PSCI test suite that sends the supported PSCI commands into TF and checks for PSCI spec compliance. This would ensure the PSCI platform porting interfaces have been implemented correctly.
3.  A test suite to drive all generic TF functionality. This would ensure all platform porting interfaces have been implemented correctly, that TF has been configured correctly, and that the generic code's assumptions are correct.
3.  A framework to allow anyone to add test cases for testing their own EL3 functionality.

Obviously some of these require a lot more effort than others.

Regards

Dan.

monkey-jsun, 2016-08-11:
Hi, Dan,

Couldn't agree more with the requirement list you give out.  I think #2 and #3 are really the key here.  #4 could evolve over time naturally instead of over-engineering at the beginning.  such a suite will not only validate the porting, but can also validate early silicon designs.  It fills in a gap between instruction-level verification and full OS (linux) validation.

Jun

danh-arm, 2016-08-11:
> but can also validate early silicon designs

I think validating silicon designs is a requirement over and above the ones I mentioned, although the other requirements would validate the hardware to some extent. For silicon design testing, most operations could be implemented on a single CPU executing entirely in the normal world, and there are other frameworks available for this (e.g. UEFI).