ARM-software/tf-issues #416: Running position independent support for arm trusted firmware ?

prime-zeng, 2016-08-11:
Since arm trusted firmware can be executed only in the fix position once it has been compiled, but we may need to a single image to run on different boards with different memory size and the memory layout is different for arm trusted firmware. 

Any plan to support or any suggestion of how to do it?

vwadekar, 2016-08-11:
I remember discussing this in the past, but the idea did not fly. I suppose
ARM did not have any plans then.

Would like to know if the stand has changed.

Thanks.

danh-arm, 2016-08-11:
This is a duplicate of the issue here: https://github.com/ARM-software/tf-issues/issues/111. There is still no plan from our side to implement this but we can raise the priority if there is enough demand. Please contact us privately if you wish to discuss our roadmap.

In terms of how to do it, that is not a simple question. It requires some serious design thinking and discussion.

prime-zeng, 2016-08-11:
@danh-arm So how to contact with you?  I think this request is a common requirement for real product.

danh-arm, 2016-08-11:
> So how to contact with you?
Initially you should contact your ARM partner manager or ARM support.

prime-zeng, 2016-08-12:
OK, got it, thinks.

closed 2016-08-12 by prime-zeng