ARM-software/tf-issues #418: What other build options should be set when we set ARM_ROTPK_LOCATION=regs

matt2048, 2016-08-19:
As it said in user-guid.md, when build FIP images with support for Trusted Board Boot, the location of the ROTPK hash must also be specified at build time via ARM_ROTPK_LOCATION option, one of the two possible values is regs, but in user-guid.md, there is an explanation for it:

regs : return the ROTPK hash stored in the Trusted root-key storage registers. The private key corresponding to this ROTPK hash is not currently available.

Does it mean that we can not use regs value for this option? If not, what other build options should be set when we set ARM_ROTPK_LOCATION=regs?

Thanks

danh-arm, 2016-08-19:
Hi @matt2048

To build TF using a specific Root of Trust Private Key (ROTPrivK), you need to specify `ROT_KEY=` in the build options. Without this option, a generated ROTPrivK is used. The ROTPK (public) hash stored in the Trusted root-key storage registers corresponds to the Juno ROTPrivK. This private key is not available outside of ARM so unfortunately the "ARM_ROTPK_LOCATION=regs" option is not currently usable outside of ARM.

The latest Foundation/Base FVPs support the ability to program the Trusted root-key storage registers during model initialization. To use that feature, the FVP platform port would need a modified implementation of `plat_get_rotpk_info() `. With that in place, it would be possible to use the "ARM_ROTPK_LOCATION=regs" option, using your own ROTPrivK. Would this feature be of use to you if we implemented it?

Note, this would not change the Juno port, since the Trusted root-key storage registers there are fixed.

Regards

Dan.

matt2048, 2016-08-22:
Hi @danh-arm 
Thanks a lot, I see. I can use ARM_ROTPK_LOCATION=devel_rsa option by now, as user-guide.md said, the ROTPK hash is hardcoded in the ARM platform port, and we can specify arm_rotprivk_rsa.pem to ROT_KEY= option. But it would be better if we can use our own ROTPrivK and put public key hash in Trusted root-key storage registers, look forward to your implementation.

Thanks

Matt.

danh-arm, 2016-09-06:
I created #423 for this. Closing the question...

closed 2016-09-06 by danh-arm