ARM-software/tf-issues #424: Potential inconsistency of psci_plat_pm_ops between primary core and secondary cores

iwishguo, 2016-09-08:
The global variable psci_plat_pm_ops is in cacheable memory and I don't find any code which will flush it to DDR. When secondary cores are powered up, the code will set the state to be MMU enabled but D-cache disabled, and in function psci_cpu_on_finish, psci_plat_pm_ops will be referenced before D-cache is enabled. So I'm afraid this will cause data inconsistency if psci_plat_pm_ops has not been evicted from cache to DDR.

Actually, we are using ATF v1.1 and have observed the issue happening. I think the issue may still exist when I review the latest code.

Could you help to confirm whether it is a bug and the best way to fix it? Right now we just add a flush statement after plat_setup_psci_ops((uintptr_t)psci_entrypoint, &psci_plat_pm_ops).

Thanks and regards.

Heyi

soby-mathew, 2016-09-08:
From my analysis, you seem to be right. We have not encountered this problem on our platforms and this may have been undetected. The dcache flush after plat_setup_psci_ops() is probably the right solution.

soby-mathew, 2016-09-09:
@iwishguo, I pushed a patch to fix this [here](https://github.com/ARM-software/arm-trusted-firmware/issues/699). Thanks for reporting this bug.

closed 2016-09-16 by davidcunado-arm