ARM-software/tf-issues #426: Question: Non Trusted Min Version Counter independence from Trusted Min Version Counter?

alex-nemirovsky-ca, 2016-09-14:
It appears that during the Min. Version Counter enforcement scan, secondary Trusted images must have the same version version number as the Root trusted node. That makes sense to me as they all reference a common Trusted Min. Version Counter.   However, it seems that Non Trusted images such as BL33, which should check a different Non-Trusted Min Version Counter, are also treat as secondary images in this scan and are bound to the same rule.  Thus if BL32 Non-Trusted Certificate has a higher version than the Trusted Certificates, then this generates an error.   Is this a bug?

danh-arm, 2016-09-15:
Hi Alex

There's no such thing as the "BL32 Non-Trusted Certificate" since BL32 is secure image, but if you mean "BL33 Non-Trusted Certificate", then you are correct this should be protected by an NV counter that is independent to the NV counter that protects the secure images. If this is not working then it would be a bug. Can you provide any more details of your environment and exactly what is going wrong?

Cheers

Dan.

alex-nemirovsky-ca, 2016-09-22:
Hi Dan,
sorry for the delay. We see an issue in our own private build. However,  I don't think the issue is platform specific, and I think it may be easily reproducible in other common environments. This is because the issue involves core code.  Allow me explain below.  

If you would like specific commit for reference, it I'm using a build based on ATF commit id "ae43c9493". 

Lets say both trusted and non-counters start at 0.  If we populate the FIP with a BL33 Cert who's Min NON-Trusted version is 1 or more higher than the version number of Trusted Certs then we hit an error in drivers/auth/auth_mod.c, function auth_nvctr() line 308.  Lets say our FIP contains certs who's trusted version 0, while the non-trusted cert is 1.

 I believe this error is hit because the Trusted counter is not being incremented, while the Non-trusted is asking for an update. However, the parent of the NON-Trusted Cert is not NULL and thus it considers it a Secondary cert.   I think its taking the flow path of line number 299, then 300, and finally 308.
: 
    292         /* Get the counter from the platform */
    293         rc = plat_get_nv_ctr(param->plat_nv_ctr->cookie, &plat_nv_ctr);
    294         return_if_error(rc);
    295 
    296         if (cert_nv_ctr < plat_nv_ctr) {
    297                 /* Invalid NV-counter */
    298                 return 1;
    299         } else if (cert_nv_ctr > plat_nv_ctr) {
    300                 if (img_desc->parent == NULL) {
    301                         /* This certificate has been signed with the ROT key.
    302                          * Update the platform counter value */
    303                         rc = plat_set_nv_ctr(param->plat_nv_ctr->cookie,
    304                                              cert_nv_ctr);
    305                         return_if_error(rc);
    306                 } else {
    307                         /* Secondary certificates cannot modify the counter */
    308                         return 1;
    309                 }
    310         }

In drivers/auth/tbbr/tbbr_cot.c, the BL32 parent is  &cot_desc[TRUSTED_KEY_CERT_ID]

 
        /*
         * Non-Trusted Firmware
         */
        [NON_TRUSTED_FW_KEY_CERT_ID] = {
                .img_id = NON_TRUSTED_FW_KEY_CERT_ID,
                .img_type = IMG_CERT,
                .parent = &cot_desc[TRUSTED_KEY_CERT_ID],
                .img_auth_methods = {
                        [0] = {
                                .type = AUTH_METHOD_SIG,
                                .param.sig = {
                                        .pk = &non_trusted_world_pk,
                                        .sig = &sig,
                                        .alg = &sig_alg,
                                        .data = &raw_data,
                                }
                        },
                        [1] = {
                                .type = AUTH_METHOD_NV_CTR,
                                .param.nv_ctr = {
                                        .cert_nv_ctr = &non_trusted_nv_ctr,
                                        .plat_nv_ctr = &non_trusted_nv_ctr
                                }
                        }
                },
                .authenticated_data = {
                        [0] = {
                                .type_desc = &nt_fw_content_pk,
                                .data = {
                                        .ptr = (void *)content_pk_buf,
                                        .len = (unsigned int)PK_DER_LEN
                                }
                        }
                }
        },

What do you think? Do you think this is really platform specific issue? 

Thanks for reviewing
Alex

danh-arm, 2016-09-22:
Thanks for the info, Alex. We think this a bug in the generic code you posted. The intent of the code is to only allow certificates signed with the ROTPrivKey to update the NV counter. This provides an additional degree of platform protection in case one of the secondary certificates incorrectly has an incremented NV counter without all the other images that are protected by the same counter having their counter value incremented. In that case, authentication fails and the platform's NV counter is not incremented. The incorrect certificate could be "rolled back" to the working version without any change needed to the other secure world images/certificates. Without this "feature", a single broken certificate could increment the platform's NV counter to its maximum value and brick the device.

A side effect of this is the Non-Trusted NV counter cannot be updated when using the TBBR CoT, since the Non-Trusted Firmware Content certificate that uses this is not signed by the ROTPrivKey. This is a bug.

A simple fix would be to remove the `if (img_desc->parent == NULL)` check in line 300 above and always update the platform NV counter. This would be TBBR compliant. However, we can see there is potentially some value in only allowing a subset of the certificates to update the platform NV counter. Fixing the problem while retaining that "feature" is a bit more work.

We will think a bit more about how much effort the latter fix would be.

alex-nemirovsky-ca, 2016-09-22:
Thanks Dan for confirming.  Of course my hesitation from removing the check on line 300, is the problem it was designed to avoid. However, OTOH, as long as the developer is using the build tools to generate the associated CERTS, then I guess that should not happen.  Only, if they avoid the stock tools and somehow update the version numbers by hand to cause a mismatch in Trusted certs version.   Is that right?

davidcunado-arm, 2016-09-22:
Internal ref: http://jira.arm.com/browse/GENFW-1604

danh-arm, 2016-09-22:
You're right that if the certs are generated using the top level Makefile, then this problem should never happen. However, in a production environment there may be different vendors for each of the secure images and they may independently generate the certs for their image. They may not even use the cert_create tool as this may not provide the level of security they need. In that case, this relies on the platform integrator to ensure that all the images / certs in the device have the correct NV counter values before attempting to deploy the combined firmware images.

So this feature would sometimes prevent the platform from being bricked in case of a mistake by one of the secure image vendors or the platform integrator. It would also prevent the platform from being bricked if one of the secondary keys was compromised and an attacker could generate a valid secure certificate with a high NV counter value and somehow deploy this image to the device.

closed 2016-12-20 by danh-arm