ARM-software/tf-issues #427: Linux 'standby' with ATF PSCI?

yonghong-ren, 2016-09-21:
"standby" and "mem" are two common Linux suspend states. "mem" is mapped to PSCI_SYSTEM_SUSPEND.  This is done in drivers/firmware/psci.c (kernel source). Is there a plan to support "standby"? I understand this is kind of outside of ATF domain. Then is there a forum for discussing Linux to PSCI interface? Thanks

sudeep-holla, 2016-09-22:
Hi yonghong-ren,

This have be discussed in past and it was concluded that there's no strong reason
for have such support in PSCI specification. Refer [1] for details. If you still believe
that you need to support standby, then you need to take up the discussion with
PSCI specification group. You can find the details for that in the PSCI specification itself.

Regards,
Sudeep

[1] http://lists.infradead.org/pipermail/linux-arm-kernel/2015-July/356467.html

danh-arm, 2016-10-12:
I'm closing this question. Any further discussion should happen in the ARM Linux kernel forum Sudeep mentioned or as feedback to the PSCI specification.

closed 2016-10-12 by danh-arm