ARM-software/tf-issues #428: CP15BEN bit setting issue for aarch32 bl33

prime-zeng, 2016-09-22:
Hi exports:
I am using aarch64 bl31 and aarch32 bl33, but i noticed that the CP15BEN of SCTLR_EL1 is not set when switching to bl33.

In the function cm_init_context_common in the lib/el3_runtime/aarch64/context_mgmt.c,
`	sctlr_elx = EP_GET_EE(ep->h.attr) ? SCTLR_EE_BIT : 0;
	if (GET_RW(ep->spsr) == MODE_RW_64)
		sctlr_elx |= SCTLR_EL1_RES1;
	else
		sctlr_elx |= SCTLR_AARCH32_EL1_RES1;
`
we can see the CP15BEN setting missing. 

And i noticed that is CP15 DBM DMB instructions is already deprecated, but in fact still used in the recent linux kernel.
 
so do we missing it here or some other reason?

Thank you.

danh-arm, 2016-09-22:
From looking at the ARM ARM (ARM DDI 0487A.j) section D7.2.81, `SCTLR_EL1.CP15BEN` has an architecturally UNKNOWN reset value. Given that the instructions this bit enables are deprecated, it seems correct for AArch64 EL3 software to default to setting this value to 0.

AArch32 EL1/EL2 software can override the setting of `SCTLR.CP15BEN` and `HSCTLR.CP15BEN`. I would say it is the responsibility of the Linux loader in AArch32 BL33 code to set these to 1 if it needs to.

Note this is in contrast to the situation when EL3 is implemented using AArch32. `SCTLR.CP15BEN` resets to 1 so it is probably correct for AArch32 EL3 software to retain the reset values for SCTLR(NS). Again, AArch32 EL1/EL2 software can override this if needed.

prime-zeng, 2016-09-23:
Looking the code bellow:
`if (GET_RW(ep->spsr) == MODE_RW_64) 
          sctlr_elx |= SCTLR_EL1_RES1;
 else 
           sctlr_elx |= SCTLR_AARCH32_EL1_RES1;
 `
when the AARCH64 EL3 knows that the EL1 is AARCH32, it will set the sctlr_el1 with SCTLR_AARCH32_EL1_RES1, so i think the SCTLR_AARCH32_EL1_RES1 should contains CP15BEN bit. and  of course, this bit can be override by the EL1 AARCH32 software, but i think SCTLR_AARCH32_EL1_RES1 also can.    
And i am using AARCH32 Linux kernel as the bl33 instead of the Linux loader, but there is no set or check of the  CP15BEN bit.

danh-arm, 2016-09-26:
Of course AArch64 EL3 _can_ set the CP15BEN bit, but the question is _should it_? Certainly SCTLR_AARCH32_EL1_RES1 should not contain this bit because it is not RES1 according to the ARM ARM.

For maximum compatibility with AArch32 EL1 code, I agree it would be better for EL3 AArch64 to set the bit. But my interpretation of the ARM ARM is the bit should _not_ be set because it has an architecturally UNKNOWN reset value and the instructions it controls are deprecated. It seems like the ARM ARM authors expect AArch32 EL1 code to take action here on systems with an AArch64 EL3.

I will seek further input from our Linux experts on what their interpretation is.

danh-arm, 2016-09-27:
I've sought advice from our Linux experts and now understand the background to this UNKNOWN reset value for `SCTLR_EL1.CP15BEN`. This is simply reflecting the different reset behaviour that is expected when EL1 is AArch64 and AArch32. When EL1 is AArch32, the reset value for `SCTLR_EL1.CP15BEN` should be 1, the same as `SCTLR.CP15BEN`.

So we will fix this in the code you refer to.

closed 2016-10-18 by danh-arm