ARM-software/tf-issues #429: Trusted Firmware v1.3 release now available, plus other info

davidcunado-arm, 2016-10-14:
Trusted Firmware version 1.3 is now available and can be found here:
https://github.com/ARM-software/arm-trusted-firmware/tree/v1.3

The headline features are:

* Support for running Trusted Firmware in the AArch32 execution state, including refactoring of the PSCI implementation into a library for integration into other EL3 Runtime Software.

* A Performance Measurement Framework (PMF) to measure the execution time of critical paths in Trusted Firmware.

* Support for PSCI_STAT_RESIDENCY, PSCI_STAT_COUNT and NODE_HW_STATE.

* Support for the ARM Cortex-A32, A35, A72 and A73 processors.

Please refer to the readme and change log [add links] for further information.

Note it is no longer necessary for contributors to send a CLA to ARM before making contributions. Contributors must instead add a "Signed-off-by:" line to each commit, which certifies that the contribution is made under the Developer Certificate of Origin (DCO).

Also note that a new security vulnerability handling process is available. This is described in the wiki:
https://github.com/ARM-software/arm-trusted-firmware/wiki/ARM-Trusted-Firmware-Security-Centre

Thanks!

prime-zeng, 2016-10-14:
Is the ARMV7 supported in AARCH32?

davidcunado-arm, 2016-10-14:
ARMv7 is not supported. 

We've not tested the AARCH32 support on ARMv7 - it's only been tested on ARMv8-A running in AArch32 execution state. 

But ARMv7 and AArch32 on ARMv8 are very similar - and so there may little or no changes required to make this work on ARMv7.

prime-zeng, 2016-10-15:
Will you claim the ARMV7 support ?

davidcunado-arm, 2016-10-17:
No, as per earlier comment. We cannot claim support as we have not validated ARM-TF on ARMv7.

lusson, 2016-10-25:
Will you support bl31 in AARCH32?

danh-arm, 2016-10-25:
> Will you support bl31 in AARCH32?

No. The AArch32 secure world architecture is fundamentally different from AArch64 and we expect all monitor functionality to be tightly integrated into the Secure Payload (BL32). The AArch32 PSCI library is provided to facilitate this. BL31 becomes an optional, transient boot image in AArch32. See this presentation for more details:
http://connect.linaro.org/resource/las16/las16-402/

closed 2016-11-23 by danh-arm