ARM-software/tf-issues #430: MDCR_EL3 not initialized in Trusted Firmware

danh-arm, 2016-10-17:
Some of the bits in the `MDCR_EL3` register have an architecturally UNKNOWN reset value. Trusted Firmware currently does not initialization this register at all, which can cause unexpected trap exceptions to EL3 on CPU implementations that do not reset these bits to zero. Cortex-A73 at least is known to be affected by this.

The EL3 architectural initialization code in Trusted Firmware should initialize this register to zero, since it does not expect to handle these trap exceptions. Similarly, TF should `MDCR_EL2` to zero, at least when starting the normal world at EL1.

Finally, `HDCR` and `SDCR` should similarly be initialized in the AArch32 EL3 case.

danh-arm, 2016-10-17:
Internal ref: http://jira.arm.com/browse/GENFW-1628

prime-zeng, 2016-11-07:
So this issue has been fixed,  where can i get the the patch?

davidcunado-arm, 2016-11-07:
I'm aiming to get the patch into a PR this week.

prime-zeng, 2016-11-08:
OK, i have met the problem already, and have a temp fix,  I'd rather to have the PR as soon as possible.

danh-arm, 2016-11-09:
Hi @prime-zeng 

Could you confirm that PR ARM-Software/arm-trusted-firmware#754 fixes this issue for you?

Thanks

prime-zeng, 2016-11-14:
i am test it on my platfrom,  only set MDCR_EL3 will work for us, and one more question:  Does the DACR32_EL2 also need to intialize in EL3, more general, do we have more reset unknow register need to be set in EL3?

danh-arm, 2016-11-14:
Thanks for testing @prime-zeng.

I don't think DACR32_EL2 needs to be intialized at EL3 since:

- EL3 has no interest in the reset value of this register, although it does save and restore it.
- EL3 operation is not affected by the reset value of this register.
- EL2 code can initialize this if it wants to (or AArch32 EL1 can initialize the corresponding DACR register).

Regarding other registers, we intend to have a look through the ARM ARM for this but haven't identified any others yet. Please let us know if you think there are any others.

closed 2016-11-18 by davidcunado-arm

davidcunado-arm, 2016-12-01:
Hi @prime-zeng 

FYI after reviewing the ARM ARM, I've created PR ARM-software/arm-trusted-firmware/pull/772 which looks to initialise any registers that have an architecturally UNKNOWN value on reset that may have an impact on system behaviour.

Thanks,

David