ARM-software/tf-issues #431: trivial comment fixes, .gitignore update

masahir0y, 2016-10-22:
Hi.

This is my first try to take part in the upstream ATF development.
Please let me know if I am doing wrong in the pull-request manner.

The following are trivial enough to be included in a single PR, I guess:
  - Ignore editor backup files.
  - Comment fixes.

If it looks OK, please pull!

The following changes since commit e1c4274091ed0861c31566f82db262d3604836a2:

  Merge pull request #735 from soby-mathew/sm/aarch32_sctlr (2016-10-17 12:20:18 +0100)

are available in the git repository at:

  git://github.com/uniphier/arm-trusted-firmware.git misc

for you to fetch changes up to 65996d27550d0c720e6c2bf6f2cea435f9c57288:

  .gitignore: ignore editor backup files (2016-10-23 02:19:01 +0900)

----------------------------------------------------------------
Masahiro Yamada (3):
      Docs: fix a typo in Porting Guide about plat_get_my_entrypoint()
      Fix comment of plat_reset_handler stub
      .gitignore: ignore editor backup files

 .gitignore                             | 1 +
 docs/porting-guide.md                  | 2 +-
 plat/common/aarch64/platform_helpers.S | 2 +-
 3 files changed, 3 insertions(+), 2 deletions(-)

danh-arm, 2016-10-24:
Hi @masahir0y

It's great you want to contribute to Trusted Firmware. To create a Pull Request for your work, follow the GitHub instructions here:
https://help.github.com/articles/creating-a-pull-request/

Make sure that the "base branch" is set to "integration" instead of "master", and follow the other contribution guidelines here:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/contributing.md

Regards

Dan.

danh-arm, 2016-11-03:
Fixed by https://github.com/ARM-software/arm-trusted-firmware/pull/742

closed 2016-11-03 by danh-arm