ARM-software/tf-issues #433: Question: LOAD_IMAGE(legacy) vs LOAD_IMAGE_V2

masahir0y, 2016-10-28:
Hi.

I am working on ATF porting for my new SoC platform.
When looking into BL1 & BL2 code, I notice two different mechanisms
in terms of image loading.
For new platform, LOAD_IMAGE_V2 := 1 is generally a preferred choice, correct?

jeenu-arm, 2016-10-28:
Yes, that's the preferred image loading scheme moving forward. You'd notice that's already enabled for AArch32 builds. This however isn't enabled for AArch64 for compatibility reasons. For a fresh port of ARM TF, I'd suggest you use the new one.

Please keep in mind that we haven't extensively tested the new scheme on AArch64 yet, but it's reasonably straight forward and is expected to work out of the box. As always, if you hit any problems, feel free to suggest fixes you find needing on the way.

danh-arm, 2016-10-31:
Also note that the TRUSTED_BOARD_BOOT build flag in combination with LOAD_IMAGE_V2 is not currently supported. We will be fixing that soon.

closed 2016-11-03 by danh-arm

nlebayon, 2016-11-24:
Hi,
I'm novice in ARM TF use, and I would like to switch our AArch32-based platform from LOAD_IMAGE to LOAD_IMAGE_V2 mode. We're working on 1.3 tag.
Are there some documents or reference code inside the tree that could help me in this task? I parsed the md files, but a more precise information would be appreciated.
Thanks in advance

sandrine-bailleux-arm, 2016-11-25:
Hi @nlebayon ,

Unfortunately, there's no transition guide to migrate from LOAD_IMAGE to LOAD_IMAGE_V2, as far as
I am aware. I you haven't already seen that, you may want to take a look at the patch that migrated the
ARM platform ports:
https://github.com/ARM-software/arm-trusted-firmware/commit/a8aa7fec1d4a6df8617c0d0463f1e10f1827a609

Sorry I'm not able to point you to a better reference.

Regards,
Sandrine

nlebayon, 2016-11-30:
Hi Sandrine,

Thanks for your feedback. 
I based my porting on the ARM platform reference.
But I see something I do not understand:
- bl2_plat_handle_post_image_load() is used when LOAD_IMAGE_V2 is enabled.
- it calls arm_get_spsr_for_bl33_entry(), which call itself plat_get_ns_image_entrypoint()
- but as said in the porting-guide.md, plat_get_ns_image_entrypoint are required only when LOAD_IMAGE_V2 is disabled. Which is inconsistent with the first bullet.

Maybe I don't have the correct interpretation of the porting guide content, but it appears that plat_get_ns_image_entrypoint is used even if LOAD_IMAGE_V2 is enabled.

Could you clarify please?

Best Regards
Nicolas

nlebayon, 2016-12-07:
Hi,

Back after a few days, we finally reach to use the LOAD_IMAGE_V2 mode inside our platform. All has been implemented under the dedicated LOAD_IMAGE_V2 flag, enabled or not inside the main Makefile.

Regarding my last message, indeed plat_get_ns_image_entrypoint has to be also enabled if LOAD_IMAGE_V2 is enabled.

Best Regards
Nicolas

danh-arm, 2016-12-07:
Hi Nicolas

You are correct that the porting guide states "Following functions are required only when LOAD_IMAGE_V2 is disabled", and then proceeds to list a number of functions including plat_get_ns_image_entrypoint(). This is a mistake and the description for that function should be moved above the previous statement.

This is a simple change and we would welcome a contribution to this effect.

Regards

Dan.