ARM-software/tf-issues #434: PSCI: psci_common.c:  error: array subscript is above array bounds

vndao, 2016-11-04:
Hi everyone,
I am having issue when building arm trusted firmware:

lib/psci/psci_common.c: In function ‘psci_do_state_coordination’:
lib/psci/psci_common.c:220:27: error: array subscript is above array bounds [-Werror=array-bounds]
  psci_req_local_pwr_states[pwrlvl - 1][cpu_idx] = req_pwr_state;
                           
Wonder if it is known issue?

By the way, i am using gcc linaro toolchain (gcc-linaro-aarch64-linuxgnu-4.8-2014.04_linux) with compiler is aarch64-linux-gnu.

Thanks,
Viet Nga

sandrine-bailleux-arm, 2016-11-04:
Hello,

I can indeed reproduce the build error you've reported using the gcc-linaro-aarch64-linux-gnu-4.8-2014.04_linux Linaro cross-toolchain. I wasn't aware of it. I suspect it might be a false positive, though, because I can't see anything wrong in the code and the index values used to access the array...

The cross-toolchain version we recommend is more recent. It is the 4.9-2015.05 Linaro one (i.e. gcc-linaro-4.9-2015.05-x86_64_aarch64-linux-gnu). You can find the link to download it from here:
https://community.arm.com/docs/DOC-10952#jive_content_id_Linaro_Release_1606

Using the 4.9 cross-toolchain, I don't get this error. Could you please give 4.9 a go and consider
upgrading to it?

Regards,
Sandrine

vndao, 2016-11-07:
Thanks Sandrine. I will upgrade my toolchain then.

closed 2016-11-07 by vndao

danh-arm, 2016-11-07:
Thanks for reporting, @vndao - it's good to be aware of this issue anyway.