ARM-software/tf-issues #435: Annotate tag instead of normal tag

masahir0y, 2016-11-10:
Unlike most other open source projects, ATF generally does not use annotate tags.
Any reason for this?

I think annotate tags are more useful, because
'git describe' is based on an annotate tag.

I see v0.2 is the last annotate tag, but it was more than three years ago.

$ git describe master 
v0.2-1534-g90d2956

sorenb-xlnx, 2016-11-10:
You can always add ```--tags``` to that commandline which should consider lightweight tags for the description.

danh-arm, 2016-11-10:
I haven't thought much about this before. We can annotate future tags if you like. I don't imagine the tag message would provide much useful information. Are you just looking for a tagger name and date?

masahir0y, 2016-11-11:
For example,  "git show v4.8" in Linux shows as follows:

The comment is quite simply "Linux 4.8".
This does not carry much information,
but I think it is meaningful by being an annotate tag.

Also, annotate tags can be accompanied with GPG signature.
(use -s option when tagging, if you have a GPG key)

With this, people will be more confident that it was tagged by you,
if you are connected with other developers by Web of Trust.

tag v4.8
Tagger: Linus Torvalds <torvalds@linux-foundation.org>
Date:   Sun Oct 2 16:24:40 2016 -0700

Linux 4.8
-----BEGIN PGP SIGNATURE-----

iQEcBAABAgAGBQJX8Zc4AAoJEHm+PkMAQRiGQG8H/2Hd4IwJh75snGY5LAiWt6ra
kGM/SobvLAMtcoxXCeHqf2bZrxa2Zz9tnEzhuLMGaf9a3l79xHa8YumK5KS1JPGV
6lZBvuPi8BIyT0cpYH000e5ehHfhP6pSGJKZ2EuLv43HcBeVZEGAf3/8jSAlNA15
bwFy2ZEkwJGThbnT6au0Y3s9h8LcNjtllu9hjfb+/9oNGvp8r4QhdVodIqIQ4cv6
SeUfv7Pn2LZDMCSaTP9bh2KaR4dwYZHFsVe75x2wND5Sfq7DVBBfFkAoV/RwJDTM
gBc3PNnmzb/tix6ohOrSQnSiGsXv1uASxvHH3RD2zG6g7Aj9Eq/+Z7ZdPu2+o+U=
=U+ef
-----END PGP SIGNATURE-----

danh-arm, 2016-11-14:
Annoted tags only seem useful to me if they are also signed. We'll sign tags in future then if that gives you more confidence in the authenticity of the tags.

danh-arm, 2016-11-23:
I'm closing this now. We'll sign future tags.

closed 2016-11-23 by danh-arm