ARM-software/tf-issues #436: Can the arm-tf support Execution state switch by a new service?

prime-zeng, 2016-11-21:
For example:
                                     SMC                                  ERET
1. NS-aarch32(EL1)-------------->aarch64(EL3)----------->NS-aarch64(EL1)
or:
                                    SMC                                   ERET 
2. NS-aarch64(EL1)-------------->aarch64(EL3)----------->NS-arch32(EL1)

arm-tf only support one NS context-->psci_context, so i am sure whether it can support my request, anyone who can help to explain?  
Thank you.

danh-arm, 2016-11-21:
Can you elaborate on your use-case here? I've seen some platforms do what you suggest during the boot sequence, to facilitate AArch64 normal world firmware dynamically selecting between AArch64 and AArch32 kernels. For that use-case, I think it would be cleaner for the device to contain both AArch32 and AArch64 versions of the normal world firmware, which the secure world dynamically selects. However, that requires some dynamic configuration support in TF that doesn't yet exist (and has been discussed elsewhere).

What you suggest is possible, but I think it would only make sense during boot, and ideally the SMC would be disabled at runtime to prevent misuse. Having 2 PSCI contexts per CPU is not going to fly. I've discussed this previously with our architects and there was no appetite for a standardised SMC interface for this. Perhaps if there was non-platform-specific implementation for this, we could discuss standardisation futher?

Regards

Dan.

prime-zeng, 2016-11-22:
My use case is in the boot sequence:
...bl31(EL3 AARCH64)------>bootloader1(EL1 AARCH32)---------->bootloader2(EL1 AARCH64)
The original question is how to boot from bootloader1 to bootloader2, and my idea is to introduce a new service for execution state switch in TF.   
And i don't what is the use case you metioned about the aarch64 to aarch32 switch,  in my case, bootloader1 is fixed to aarch32 and very hard to change to aarch64, that is why i have the request.

danh-arm, 2016-11-22:
OK, I understand your use-case, thanks. The problem with this is that the proposed service would need to overwrite the normal world context for the primary CPU after this has been initialized in the PSCI implementation.  So this would probably need to be described in the PSCI specification. It would easier to implement if the proposed SMC was only available before any CPU_ON calls. I will raise this again with our architects.

prime-zeng, 2016-11-22:
In my case, from bootloader1 to bootloader2,  only the boot cpu is on and the CPU_ON calls are not called .

hzhuang1, 2016-11-22:
@prime-zeng You said that it was just in bootstage and only the primary cpu was on. Does primary cpu boot as aarch32? Could bootloader1 run in EL3 mode before BL31 initialized?

prime-zeng, 2016-11-23:
@hzhuang1 , no, Bootload1 must run in NON-Secure state, so it can't run in EL3 mode.

danh-arm, 2016-12-15:
I've had feedback from our architects that this is unlikely to ever be defined in the PSCI specification, mainly becuuse it's unlikely to ever be used by generic OS code. However, I can see there is potential value in providing a reference implementation for this. Most of the implementation would exist in generic code with a thin platform specific layer to invoke this (e.g. as a platform SiP service).

I'll add this as a task to our backlog.

danh-arm, 2016-12-15:
Internal ref: http://jira.arm.com/browse/GENFW-1713

prime-zeng, 2016-12-16:
@danh-arm  think you, so you will provide a reference implementation and how can get it?  In fact, i have tried to implement it as a new service, and what i am worried about is that the new service will break the psci design.

danh-arm, 2016-12-19:
There will definitely need to be changes to the generic PSCI code to support this, but I think it should be possible to make this an optional part of the PSCI implementation without breaking the overall design or compliance to the spec.

Yes, we intend to do this but we haven't planned it in yet.

closed 2017-05-11 by davidcunado-arm