ARM-software/tf-issues #438: Build failure when using GCC6

omnium21, 2016-12-05:
ARM-TF fails to build with GCC6.

I used Linaro GCC releases from 16.08 (GCC6.1) and 16.11 (GCC6.2) and both compilers show the same problem.  Here's Linaro's 16.11 GCC release:

http://releases.linaro.org/components/toolchain/binaries/6.2-2016.11/

And here's the set of error messages I get:

  CC      services/spd/opteed/opteed_main.c
services/spd/opteed/opteed_main.c: In function 'opteed_smc_handler':
services/spd/opteed/opteed_main.c:297:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_ENTRY_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:344:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_ON_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:345:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_RESUME_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:356:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_OFF_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:357:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_SUSPEND_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:358:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_SYSTEM_OFF_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:359:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_SYSTEM_RESET_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:373:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_CALL_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:397:2: error: case label is not an
integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_FIQ_DONE:
  ^~~~
  CC      services/spd/opteed/opteed_pm.c

sandrine-bailleux-arm, 2016-12-05:
Hello Ryan,

From my initial investigation, the build failure you reported seems to be due to the OPTEE SPD
code relying on undefined behaviour in C. I was able to reproduce the same build failure using
the following piece of code:

```
void foo(unsigned int a)
{
    switch (a) {
    case (1 << 31):
        break;
    }
}
```

```
test.c: In function 'foo':
test.c:6:5: warning: case label is not an integer constant expression [-Wpedantic]
     case (1 << 31):
     ^~~~
```

From what I understood of the C specification, this is undefined behaviour. See section 6.5.7 "Bitwise
shift operators" in the C99 spec, I believe this is falling under the case of 1 being a signed type, non-negative value BUT (1 * 2e31) not being representable in the result type (i.e. int in our case). 

In my sample code above, if I change `1 << 31` into `1 << 30`, (1 * 2e30) is representable in the
result type and so I get not error. Similarly, if I explicitly qualify the constant 1 as unsigned, i.e. using
`(1U << 31)`, then the compiler is happy.

So I guess we need a patch that adds the 'U' suffix to some of the constants defined in
`teesmc_opteed.h` and used from opteed_main.c.

omnium21, 2016-12-05:
Hi Sandrine,
Well spotted! I just tried your idea by adding U to the end of the constants and now I get these warnings instead:

  CC      services/spd/opteed/opteed_main.c
services/spd/opteed/opteed_main.c: In function 'opteed_smc_handler':
services/spd/opteed/opteed_main.c:297:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_ENTRY_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:344:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_ON_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:345:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_RESUME_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:356:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_OFF_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:357:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_SUSPEND_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:358:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_SYSTEM_OFF_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:359:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_SYSTEM_RESET_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:373:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_CALL_DONE:
  ^~~~
services/spd/opteed/opteed_main.c:397:2: error: case label is not an integer constant expression [-Werror=pedantic]
  case TEESMC_OPTEED_RETURN_FIQ_DONE:
  ^~~~

omnium21, 2016-12-05:
Hi Sandrine,
Well spotted. I only had to change the SMC_32 and SMC_TYPE_FAST constants in smcc.h to get it to build without warnings as they are the ones shifted in the TEESMC_OPTEED_RV macro.  Although a proper fix ought to apply to the other constants too, I guess.  And maybe the magic 62 (which should maybe be a constant with a meaningful name?).

jbech-linaro, 2016-12-06:
@danh-arm, @sandrine-bailleux-arm do you want a fix from us or is this something you can provide on your own? Please let me know what you prefer. (FYI, we saw the same issue in the [OP-TEE driver a while ago](https://github.com/linaro-swg/linux/commit/7fa0f28a7ec5c0cc77edc41e16b11f01b4624f6b) too)

davidcunado-arm, 2016-12-12:
Internal ref: http://jira.arm.com/browse/GENFW-1704

closed 2017-02-02 by danh-arm