ARM-software/tf-issues #441: Default EL3 interrupts registration

chinnabha, 2016-12-15:
Hi Achin Gupta/Soby Mathew,

Couple of questions w.r.t. EL3 interrupts. Please help.

1. Why is the arm gicv3 driver configured to route SGIs 8,14 to the CPU (g0_interrupt_array[ ] = {8, 14})? Any specific reason to made those two SGIs default?
2. Why the corresponding interrupt handlers not registered?

soby-mathew, 2016-12-15:
> Why is the arm gicv3 driver configured to route SGIs 8,14 to the CPU (g0_interrupt_array[ ] = {8, 14})? Any specific reason to made those two SGIs default?

The 2 SGIs were chosen arbitrarily to validate the GICv3 driver interface. There was no specific reason to assign these 2 SGIs as G0 interrupts AFAIU.

> Why the corresponding interrupt handlers not registered?

The Interrupt management framework in Trusted Firmware allows to register EL3 interrupts. But handling of EL3 interrupts need careful analysis into prioritization of these interrupts wrt to other secure interrupts, routing strategies, pre-emption mechanisms when other lower priority S-EL1 interrupts are in flight etc. There is some analysis ongoing internally to evolve a framework to generically handle the complexities, but till that work is done, it is upto the Secure Payload Dispatchers (SPDs) to decide how to make use of EL3 interrupts for its specific use case.

Hope that helps.

chinnabha, 2016-12-15:
@soby-mathew,

Sure, we can wait for the evolution of the framework. Thanks for the help in answering the queries.

chinnabha, 2016-12-16:
Hi Soby-mathew, 

overall, it seems EL3 runtime services can register 3 top level interrupt handlers to route the incoming interrupts from GICv3:

1.  To handle the EL3 interrupts in BL31 runtime services  
2.  To route the S-EL1/EL0 interrupts when generated during code execution in the non-secure state (NS-EL2/EL1/EL0).
3.  To route NS-EL2/EL1/EL0 when generated during code executing in S-EL1/EL0.

As per your earlier response the interrupt handler registration for case 1 is not present in EL3 runtime services as the framework is still under investigation. The TSPD secure pay load dispatcher has handlers registered for cases 2 and 3. 

Where as, the opteed secure play load dispatcher has handler registered to cover case 2 alone. It is not covering case 3. How and where does the optee handles case 3? Does it routes the received  NS-EL2/EL1/EL0 interrupts directly from S-EL1 itself to Non-secure software without first reaching the opteed secure payload dispatcher?  

Though this may be a question to optee guys, you might be aware of this scenario. Please help.

soby-mathew, 2016-12-16:
It is not necessary for SPDs to support NS Interrupt pre-emption of `standard SMC` execution (case 3). The TSP implements this to demonstrate the capability and to provide reference implementation. I guess it depends on the Optee to decide whether they deem it useful to support this or not.

> Does it routes the received NS-EL2/EL1/EL0 interrupts directly from S-EL1 to Non-secure software without first reaching the opteed dispatcher?

I don't think Optee would bypass Dispatcher when switching from Secure to Normal world.

chinnabha, 2016-12-16:
Thanks much @soby-mathew for the prompt help. This is very helpful.

Hi Jens,

Could you please pitch in and confirm whether the optee handles the case 3?

chinnabha, 2016-12-18:
Ok, I have gone through the optee code. As you said Optee doesn't have a mechanism to bypass the EL3 runtime services dispatcher to support case 3 in aarch64. As per my understanding it supports case 2 only.

chinnabha, 2016-12-22:
Hi @soby-mathew, Can you please tell me how you have validated the scenario 2 above? The default G1S interrupts configured in atf are g1s_interrupt_array[] = {29, 8, 9, 10, 11, 12, 13, 14, 15}. Out of those nine interrupts eight are secure SGIs. Non-secure software cannot generate the secure SGIs via writing to ICC_SGI1R_EL1. That leave us with only interrupt 29. To validate scenario 2, we need to create a scenario wherein interrupt 29 is raised when the processor is in non-secure software execution.

Assuming that there is no real hardware to generate interrupts (for example hardware corresponds to S-PPI 29), is there any way to generate a secure PPI/SPI interrupt using software?

davidcunado-arm, 2016-12-22:
@chinnabha apologies but due to national holidays in UK, we'll respond early in January - thanks!

chinnabha, 2016-12-22:
@davidcunado-arm, that is perfectly ok, thanks for the update.

soby-mathew, 2017-01-03:
> To validate scenario 2, we need to create a scenario wherein interrupt 29 is raised when the processor is in non-secure software execution.

Yes, this is exactly what we do to validate scenario 2. The interrupt 29 corresponds to the the secure generic timer interrupt on ARM Standard platforms which is configured by TSP during initialization. The relevant initialization code can be found [here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl32/tsp/tsp_main.c#L114).

chinnabha, 2017-01-06:
That looks a lot tedious process, and the only way to test secure PPIs.

But for secure SPIs there seems a simpler solution exist:
GICD_ISPENDR/GICD_ICPENDR together with GICD_NSACR. The GICD_NSACR register (with 2-bit field value for each SPI = 0b11) permit Non-secure software to generate secure SPI interrupts.

chinnabha, 2017-01-06:
@soby-mathew, Can you please point me to the bl31 code, where we clear the I and F mask bits of the PSTATE to receive the IRQs/FIQs by the EL3 runtime services?

For me it seems the runtime services clear DAIF_ABT_BIT alone.

soby-mathew, 2017-01-09:
BL31 is non-premptible and it doesn't clear I and F masks during runtime. The pending EL3 interrupts will remain pending as long as the CPU is executing at EL3 and when the CPU transitions to a lower exception level, the configured Interrupt routing will cause the CPU to switch back to EL3 and the interrupts will get handled.

chinnabha, 2017-01-09:
I don't understand how the EL3 can ever handle the (I and F type) interrupts if the I and F mask bits are not cleared in its PSTATE. I am not bothered much about EL3 interrupts. My worry is for S-EL1 interrupts handling via **"opteed_sel1_interrupt_handler( )"**.

opteed_sel1_interrupt_handler( ) is supposed to be the entry point (from the EL3 interrupt vector) for the S-EL1/EL0 interrupts if those get triggered during code execution in the non-secure state. Non-secure software (NS-EL2/EL1/EL0) doesn't even aware that there happened a secure interrupt. Now please explain me how the processor reaches opteed_sel1_interrupt_handler( ) from non-secure world, if I and F masks bits are not cleared in EL3 PSTATE.

soby-mathew, 2017-01-10:
The PSTATE.I and F flags only mask interrupts when the target exception level is the current exception level. Please refer to section D1.14.1 and D.14.2 of [ARMv8 ARM](https://silver.arm.com/download/download.tm?pv=3742010&p=1343131) for more details on "Asynchronous exception routing" and "Asynchrounous exception masking".

chinnabha, 2017-01-11:
This is really helpful. Thanks much. 

The ATF sets SCR_EL3.IRQ and SCR_EL3.FIQ to zero, which implies irrespective of the SCR_EL3.RW and HCR_EL2.{AMO, IMO, FMO} values zero or one, the target exception level is always either EL1 or EL2 or P. It is never going to be EL3. 

That essentially implies the scenario that I referred above can never able to invoke opteed_sel1_interrupt_handler( ) to route the S-EL1 interrupts via EL3. This is a confusion. The ATF registered opteed_sel1_interrupt_handler( ), but it doesn't program SCR_EL3.IRQ and SCR_EL3.FIQ to 1 to invoke that interrupt. Is this intentionally left for platform specific handling? Does those who wants this routing need to set SCR_EL3.IRQ = SCR_EL3.FIQ = 1?

soby-mathew, 2017-01-11:
> The ATF registered opteed_sel1_interrupt_handler( ), but it doesn't program SCR_EL3.IRQ and SCR_EL3.FIQ to 1 to invoke that interrupt. 

No, ATF provides the interrupt management framework which does the appropriate programming of SCR_EL3.IRQ and SCR_EL3.FIQ depending on the routing model. The [`set_routing_model()`](https://github.com/ARM-software/arm-trusted-firmware/blob/949a52d24eea48a58608645b6536ab7158abcbbb/bl31/interrupt_mgmt.c#L146) determines the SCR routing bits during [interrupt registration](https://github.com/ARM-software/arm-trusted-firmware/blob/949a52d24eea48a58608645b6536ab7158abcbbb/bl31/interrupt_mgmt.c#L210). These SCR routing bits are programmed during secure/non-secure world switch.  The routing model is determined via the routing `flags` which are set using the `set_interrupt_rm_flag()` function.

For more details on interrupt management framework and routing models, please refer [here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/interrupt-framework-design.md).

chinnabha, 2017-01-14:
Thanks again @soby-mathew for your help. I will revisit this thread after spending decent time on this.

closed 2017-01-14 by chinnabha