ARM-software/tf-issues #442: Intersection of interrupts across Grp0, S-Grp1, and NS-Grp1 interrupts.

chinnabha, 2016-12-15:
Hi Sobby-mathew/Achin Gupta,

Can there exist an intersection while configuring a specific interrupt as one of the three types? For example, can I include an interrupt 'X' both in g1s and g0 interrupt arrays?
g1s_interrupt_array[ ] = { X } and 
g0_interrupt_array[ ] = { X }   

Or, every interrupt in the system needs to falls into only one category: g0, or g1s or g1ns?

soby-mathew, 2016-12-16:
> Or, every interrupt in the system needs to falls into only one category: g0, or g1s or g1ns?

Yes, an interrupt must be categorized into one of the Interrupt groups. You cannot have an interrupt in 2 groups.

chinnabha, 2016-12-16:
Ok, thanks for the help.

closed 2016-12-16 by chinnabha