ARM-software/tf-issues #444: Aarch64 system register definitions

chinnabha, 2016-12-18:
Hi,

Can someone point me to the file wherein the system register read/write definitions are present?
When I grep for them I am unable to figure out their definitions. Seems those are integrated into some binary.  I am unaware of that file.

read_scr_el3()
write_scr_el3()
write_icc_sre_el3()
read_icc_sre_el3()
read_icc_hppiro_el1()
read_icc_hppir1_el1()

sorenb-xlnx, 2016-12-18:
For aarch64 those are in https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/lib/aarch64/arch_helpers.h, the 32-bit version is https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/lib/aarch32/arch_helpers.h. They are implemented as macros that might make grepping fail.

chinnabha, 2016-12-18:
Thanks to you @sorenb-xlnx for the help.

closed 2016-12-18 by chinnabha