ARM-software/tf-issues #446: Inconsistency between #if and #ifdef

masahir0y, 2016-12-26:
In ATF, apparently, there are two groups for boolean macros.
[1] Always defined, as 1 or 0.
[2] Defined as 1, or undefined

For [1], #if FOO or #if !FOO must be used because #ifdef FOO is always true.
Options passed by $(call add_define, ...) is are the cases.

For [2], #ifdef FOO or #ifndef FOO is the correct syntax.

I am sometime confused.  And even maintainers are confused, I guess.
I sent pull-request #800 to fix some I stopped by.

But, in the first place, I do not understand why ATF chose such design.
Is there a reason for this?  Should we fix the inconsistency? 

This is nasty, because I need to confirm how the macro is defined
when I write code (and error-prune).

soby-mathew, 2017-01-03:
@masahir0y,
Thanks for your pull request. AIU, if a build macro is undefined, it evaluates to 0. Hence #if FOO, when FOO is undefined would evaluate to #if 0. This achieves the intention. As you say, perhaps it is less confusing if we use #ifdef FOO, 

We would always have 2 groups of build macros and we cannot always follow [1] as some build macros are meaningful only for some BL image / build architecture / platform. We hope to review your PR next week and perhaps amend the coding guidelines accordingly.

danh-arm, 2017-01-11:
I think the different sytles have evolved simply due to individual programmer's preference. I agree the variation is confusing and we should stick to one style. So I approve this change in principle.

danh-arm, 2017-01-26:
After thinking about this issue some more, I think the source of the inconsistency is in the `add_define` make function in `build_macros.mk`. As the comment says, this does `-DFOO if $(FOO) is empty; -DFOO=$(FOO) otherwise`, which can result in the 2 different types of boolean macros mentioned.

To make the codebase more consistent I can see 3 solutions:
1. Just use `#if` instead of `#ifdef` everywhere. This would "work" for the reason @soby-mathew states but is "weird" as @masahir0y states. Also this would revert @masahir0y's recent changes.

2. Change `add_define` to only do `-DFOO=$(FOO)` (throwing an error if undefined). Add a new function (e.g. `add_bool_define`) which only does `-DFOO`. Update all code to use `add_boot_define` and `#ifdef` where appropriate. This change is obviously much more pervasive and may cause issues for non-upstream platforms (though they should be easy to fix).

3. Do 2 but provide some migration path by not changing add_define (or somehow making this configurable).

Any thoughts?

masahir0y, 2017-01-28:
> Just use #if instead of #ifdef everywhere. This would "work" for the reason @soby-mathew states

*By default*.

You can use -Wundef to detect wrong usage of #if.

    -Wundef
     Warn whenever an identifier which is not a macro is encountered in an ‘#if’ directive,
     outside of ‘defined’. Such identifiers are replaced with zero. 

For example, Linux Kernel adds -Wundef to CFLAGS in order to shut out silly code.

We can make it work by not giving -Wundef, but this approach looks like we are hiding issues.

2  (or 3) seems appropriate in a long run.   

Programmers who came from other projects tend to assume
boolean macros are defined or undefined  ( instead of 1 or 0).

At first, I made a mistake in the following
https://github.com/ARM-software/arm-trusted-firmware/pull/789
where I used #ifndef DEBUG instead of #if !DEBUG

The choice between 2 and 3 depends on
how we should be tolerant for non-upstreamed code.

danh-arm, 2017-01-30:
Thanks for the `-Wundef` tip. I agree 2 or 3 is the better way to go.

We like to maintain source compatibility where possible. I think it would be slightly friendlier to implement 3 by emitting a make warning when add_define is called with an undefined variable (e.g. "Calling add_define with an undefined variable is deprecated"). We can also add `-Wundef` to `TF_CFLAGS` when ERROR_DEPRECATED=1. When we remove this deprecated functionality, we can change the make warning to an error.

We can use this issue to track progress of this change.

danh-arm, 2017-01-30:
Internal ref: https://jira.arm.com/browse/GENFW-1750