ARM-software/tf-issues #448: Question: How much memory is required to enable Trusted Board Boot?

masahir0y, 2017-01-06:
Hi.

I am poring ATF for my SoC.
My SoC only has 64KB SRAM.

Without TRUSTED_BOARD_BOOT, it is working fine.
I only needed 24KB for basic SoC setup, DRAM initialization,
console support, and I/O framework support provided by the drivers/io/ directory.
I still had approximately 40KB room left.

However, when I enabled TRUSTED_BOARD_BOOT, it exceeded my 64KB limit.

What I did was:

 - Download  mbedtls-2.4.0 libary
 - Include  drivers/auth/mbedtls/mbedtls_crypto.mk  from my own platform.mk
 - Compile drivers/auth/{auth_mod.c,crypto_mod.c,tbbr/tbbr_cot.c} and implement some plat_* callbacks.

Is it difficult to enable TBB in my case?
How much memory do you generally expect for ARM standard boot flow,
staring from BL1, verifying images with TBB.

soby-mathew, 2017-01-17:
@masahir0y, the memory usage for TBBR is significant. Let me do an analysis and get back to you on this.

soby-mathew, 2017-01-31:
@masahir0y, apologize for the delay regarding this query. I finally had some time to look at this. I compiled the Trusted Firmware code for FVP platform at SHA [f38d93f](https://github.com/ARM-software/arm-trusted-firmware/tree/f38d93fdbf82ebec321c70f7bc15fe12b7d57275) and my findings are given below.

BL1 Section | Size increase with TBBR
---------------- | ---------------------------------
.text | 35 KB
.ro_data | 14.4 KB
.data | 1.03 KB
stacks | 2.93 KB
bss | 9.4 KB

BL2 Section | Size increase with TBBR
---------------- | ---------------------------------
.text | 32 KB
.ro_data | 12.28 KB
.data | < 1 KB
stacks | 3 KB
bss | 9.4 KB

I have tried to remove effects of alignment requirement of the sections. So it looks like the static binary image increases by around 50 KB (text + ro + data) and runtime memory footprint increases by 62 KB for BL1 with TBBR. The size increases are similar for BL2 as well. Depending on how the images are overlaid at runtime, you are looking at a worst case runtime memory footprint increase of 120 KB for both BL1 and BL2 combined when TBBR is enabled.

danh-arm, 2017-02-01:
Thanks to @soby-mathew for the numbers. I should just qualify this by saying there are several known RAM optimizations that could be made to reduce peak RAM usage; we just haven't prioritized implementing them yet. For example:

*   Merging BL1/BL2 functionality (as discussed in #445) for systems that have "pseudo-ROM" BL1.
*   Adding support for BL2 to call into a BL1 shared library, containing for example the crypto libraries, which are the main cause of the code bloat.
*   Putting cold boot and runtime code/data in separate sections (for BL31) and reclaiming the cold boot code/data at runtime.

...

masahir0y, 2017-02-03:
Thanks @soby-mathew .  I had got similar numbers for my platform too.

The three ideas suggested by Dan seem to be useful to cut the duplicated code across images (so, will decrease the sum of memory footprint from all images), but does not solve my problem because my 64KB SRAM cannot accommodate even a single image if TBB is enabled.  This is apparent from Soby's numbers; the TBB will increase the single image (BL1 or BL2) by 50KB.

My first plan was like this

[1] Real ROM in our SoC  (this authenticates [2] in our own crypto code)
[2] pseudo-ROM BL1 runs on SRAM (this authenticates [3] by ATF standard crypto method)
[3] BL2 runs on DRAM (this authenticates rest of images by ATF standard crypto method)
[4] BL31 runs on DRAM
    ...

The problem is pseudo-ROM BL1 does not fit in 64KB SRAM if TBB is enabled.
After fiddling with this for a while, I decided this is impossible.

I guess the root cause of  the problem might be too luxury TBBR design.  I wonder if parsing X.509  certificates was really needed.  (Because I am not a security export, I may be missing something...)
This was my point of my question subject "How much memory is required to enable Trusted Board Boot?"   Actually, I had noticed the answer is bigger than 64KB.
My question implies "Is the amount of required memory reasonable?"

SoC vendors are so pressed for a competitive business in markets of phones, TVs, etc.   We are not selling crazily expensive boards like Juno.  Huge SRAM can not be expected in real products, I think.
I see upstreamed code from Mediatek, Rockchip, NVIDIA, Xilinx.  They all use only BL31.
BL1 and BL2 are never used by any SoC vendors.  I am curious about the reason.

The docs/auth-framework.md says the framework has been designed in a modular approach.
So, a simpler implementation might be possible, but I am not sure.

The solution I finally got after long struggle is like this:

Compress BL1 image (= pseudo-ROM) with GZIP.
I implemented my own firmware with GZIP-decompressor and SoC init code, piggybacking the compressed BL1.  This image fits within 64KB.   Run it on SRAM, and  expand BL1 onto DRAM.  How crazy! But works.

danh-arm, 2017-02-06:
Hi @masahir0y. 

You're right that software based crypto libraries and X509 parsing is expensive in RAM. Our initial intention was to provide a functional and flexible TBBR implementation; an optimal solution is yet to come. As you mentioned, the authentication framework allows for other crypto and image parsing modules.

Another improvement I forgot to mention before is to make use of hardware crypto functionality, either by implementing support for this under the existing mbed TLS APIs or directly as a new TF crypto module. This would improve both performance and RAM usage. Alternatively, I wonder if the crypto mechanism in your real ROM could be re-used in TF using a custom crypto module?

As you've observed, most vendors are using BL31 but many have their own solutions for BL1/BL2. Hence most of our efforts have been focussed on BL31. However, (with community help) we'd like to improvement all parts of ARM TF.

Regards

Dan.

masahir0y, 2017-02-07:
> Another improvement I forgot to mention before is to make use of hardware crypto functionality, either by implementing support for this under the existing mbed TLS APIs or directly as a new TF crypto module. This would improve both performance and RAM usage. 

Right.  Sales guys from ARM Ltd. mentioned hardware crypto engines.
It may happen in our future SoCs,  but I have nothing to do for existing SoCs.

> Alternatively, I wonder if the crypto mechanism in your real ROM could be re-used in TF using a custom crypto module?

Could be, but the implementation in the real ROM is not very nice.  It is so hard-coded that only works with the Root-of-Trust key embedded in the ROM.  It can not achieve Trust-of-chain.

So, I am motivated to use the standard TBBR of BL1 and BL2.

closed 2017-07-06 by danh-arm