ARM-software/tf-issues #449: BL31 is enormously huge for rk3399

AlexanderAmelkin, 2017-01-20:
If I build bl31 as follows using gcc-linaro-6.2.1-2016.11-x86_64_aarch64-linux-gnu:
```
make CROSS_COMPILE=aarch64-linux-gnu- PLAT=rk3399 bl31
```

I get the following result:
```
$ ls -la build/rk3399/release/
total 172
drwxr-xr-x 4 amelkin amelkin       4096 янв 20 15:10 .
drwxr-xr-x 3 amelkin amelkin       4096 янв 20 15:10 ..
drwxr-xr-x 2 amelkin amelkin       4096 янв 20 15:10 bl31
-rwxr-xr-x 1 amelkin amelkin 4287315968 янв 20 15:10 bl31.bin
drwxr-xr-x 2 amelkin amelkin       4096 янв 20 15:10 m0
```

As you can see, the `bl31.bin` file is 4 gigs large! Obviously, a loader of that size can not be used.
I believe this happens because of `SRAM` memory region defined in `plat/rockchip/rk3399/include/plat.ld.S` origin is set to `SRAM_BASE`, which is `MMIO_BASE+0x18c0000`, which evaluates to `0xfe000000 + 0x18c0000`.

Somehow, the rk3399bl31.bin file supplied by Rockchip in binary form is just 356 KiB large.

antonio-nino-diaz-arm, 2017-01-20:
You are right, the problem is that the .bin file represents the structure of all regions included in the .elf file and, as SRAM_BASE is a number close to 4GiB, the final .bin file has this huge size. I suppose you are talking about this `rk3399bl31.bin` file when you say "the file supplied by Rockchip", right? https://github.com/rockchip-linux/rkbin/blob/master/rk33/rk3399bl31.bin

in `plat/rockchip/rk3399/include/plat.ld.S` there are  3 sections. They are using `.sram.data` to store a struct of the file `plat/rockchip/rk3399/drivers/dram/dram.c`, `.sram.text` for a lot of functions located in the file `plat/rockchip/rk3399/drivers/dram/suspend.c`, and `.sram.incbin` for the M0 binary, as you can see in the file `plat/rockchip/rk3399/drivers/pmu/pmu_fw.c`.

As far as I see, they aren't loading any of them using the TF, they are just using the TF to prepare the translation tables for `.text_sram` and `.data_sram`. It's likely that they are using their own tools to generate a custom binary file from the .elf file and their own miniloader that knows about the structure of the sections in this custom .bin file and loads them before jumping to the start of BL31.

Please, @Caesar-github, could you help us?

jwerner-chromium, 2017-01-20:
We've already had a discussion about this with @danh-arm (is this how you CC people on Github?) over email. I think our consensus was that the current ARM TF requirement of always generating a raw .bin file doesn't work for platforms like rk3399 which require multiple distinct program segments in BL31, and that we should therefore add a Makefile option to prevent generating that final .bin file for certain platforms. I guess we didn't quite clarify who would be making that change, though. ;)

I think the only loader that the rk3399 port works with right now is coreboot, which takes the .elf directly and converts it into its own loadable object format which supports multiple segments. I'm not sure if Rockchip has any plans of ever supporting ARM's BL2. If they do, its loading mechanism would have to be expanded to also support multiple segments.

danh-arm, 2017-01-23:
Thanks for jumping in here, Julius. Yes, I agree that the final .bin file doesn't make sense for some platforms so a build option to prevent building this would be good. We just need "someone" to volunteer to make this change.

In the longer term we want to look again at the binary and FIP structures to be more flexible/standard and support use-cases like multiple section loading, ELF format, image meta-data, etc.. We're hoping to do some more investigation on this then propose some in this forum (unless others propose something first).

AlexanderAmelkin, 2017-02-01:
I know this is not your problem, but they say here to use bl31.bin:
https://github.com/rockchip-linux/u-boot/board/rockchip/evb_rk3399/README
That is how I discovered that bl31.bin for RK3399 is huge. I should probably file an issue with their github repo and point them here.

antonio-nino-diaz-arm, 2017-02-01:
That file was last edited 6 months ago, the problem with the bl31.bin appeared just a few months ago (with this commit, if I'm not mistaken: https://github.com/ARM-software/arm-trusted-firmware/commit/8382e17c4c6bffd15119dfce1ee4372e3c1a7890). That readme is probably just outdated.

danh-arm, 2017-02-01:
As @jwerner-chromium says, it's unlikely that u-boot currently supports unpacking the multiple sections of the RK3399 bl31.elf. So this is probably more than just a documentation issue.

AlexanderAmelkin, 2017-02-03:
@danh-arm, AFAIU, that documentation describes how to prepare the loaders that will load U-Boot, not vice versa. So U-boot has nothing to do with mutiple section ELF format of BL31. Rockchip uses some other initial miniloader that is intended to boot Trusted Firmware. U-Boot is the last one in chain.

danh-arm, 2017-02-06:
OK @AlexanderAmelkin. Thanks for the clarification. I've seen other platforms use U-Boot SPL to load TF BL31 and U-Boot proper, but I guess that's not the case here.

vagrantc, 2018-04-02:
This same issue appears to affect the build with ATF 1.5 as well with plat=qemu, so it isn't rk3399 specific.

jwerner-chromium, 2018-04-02:
The reason this happens on RK3399 is because it's using a custom linker script to define a separate segment that is far away from the main BL31 segments. Looks like plat/qemu isn't defining PLAT_EXTRA_LD_SCRIPT at the moment, so it can't really be doing the same thing. You may have some problem with your binary size, but it's probably not this one so I would open a new issue for it. Attaching the output of objdump -ph bl31.elf might help narrow it down.

SiennaStellar, 2018-05-13:
Hi guys, did anyone solve this issue?

antonio-nino-diaz-arm, 2019-04-24:
This proposal would fix the problem:

https://review.trustedfirmware.org/#/c/TF-A/trusted-firmware-a/+/924/
https://review.trustedfirmware.org/#/c/TF-A/trusted-firmware-a/+/923/

antonio-nino-diaz-arm, 2019-05-03:
Fixed in:
https://review.trustedfirmware.org/#/c/TF-A/trusted-firmware-a/+/924/
https://review.trustedfirmware.org/#/c/TF-A/trusted-firmware-a/+/923/

closed 2019-05-03 by antonio-nino-diaz-arm