ARM-software/tf-issues #452: [Tegra] DSTREAM failure during bl31.elf loading

vwadekar, 2017-01-26:
When we try to load the bl31.elf (DEBUG mode) using DSTREAM-5, we see the following error message.

---
ERROR(CMD685-IMG54): 
! Failed to load symbols for "bl31.elf"
! Failed to demand load DWARF debugging information: section .debug_info, offset 0xb
----

The base makefile enables DWARF-2 for DEBUG mode, but still the debug_info section seems to have problems.

GCC version: aarch64-linux-gnu_4.9-2015.05

soby-mathew, 2017-01-28:
@vwadekar ,  I am able to load the bl31.elf symbols in DS-5 for the FVP platform without any issues using `add-symbol-file` command. Hence I don't suspect the sections being included in the elf binary.

vwadekar, 2017-01-31:
Thanks Soby. This might be a problem with the Tegra port.

Can you please help me debug this? How should I resolve this problem?

soby-mathew, 2017-02-01:
The Tegra port uses the same LD script as FVP and hence I don't suspect the Tegra port as such. Would be worth to try if you are able to load symbols for FVP and I suspect the problem could be down to DS-5 / DStream configuration. The DS-5/Dstream support will be able to help you more in this regard.

jeenu-arm, 2017-02-01:
Some suggestions, if you haven't tried them already:

* Has this issue suddenly appeared with a specific update to ATF? If so it'd be good to bisect and identify the specific commit
* Have you tried a different, perhaps recent, tool chain?
* Is this a problem only with the BL31 image? Does other images also have this problem?
* The command I typically use is `add-symbol-file /path/to/bl31.elf EL3:0x0`. Perhaps you want to try a variant of this.

vwadekar, 2017-02-01:
Thanks for the suggestions. I will try them out.

closed 2017-07-06 by danh-arm