ARM-software/tf-issues #455: MISRA/Coverity compliance

vwadekar, 2017-02-01:
ARM Trusted Firmware goes or will go into many markets. Some of these markets have stringent safety requirements. Being MISRA compliant is one of the requirements. Coverity clean might be another.

Are there plans to get the code through the MISRA and Coverity scans? We are already cleaning up the Tegra code and would start pushing the fixes upstream. But don't know the plans for the non-platform code.

danh-arm, 2017-02-02:
Hi @vwadekar 

We already regularly run Coverity online:
https://scan.coverity.com/projects/arm-software-arm-trusted-firmware

This covers all upstream code (except files that are not included in at least 1 build configuration).

We're also looking into other static checkers (e.g. UBSAN, cppcheck). Currently we don't have much knowledge of the MISRA standards although we understand this is becoming increasingly important. Do you have any advice on tools that can help here?

Regards

Dan.

vwadekar, 2017-02-03:
@danh-arm

Fortunately, you only need the Coverity license to start MISRA scanning. We
internally use v8.7.0, so it would be great if you start with the same one.

On running the MISRA scan you will get a report with defects marked as
"Mandatory", "Required" and "Advisory". The rule of thumb (that we use) is
to fix all the "Mandatory" and "Required" defects. The "Advisory" defects
are good to have. You can plan to fix them as well to be MISRA clean.

As always, we can help you fix the defects in the common code so that you
don't do all the heavy lifting. But it would really help if we have MISRA
scans at the source as well, so that no new defects are introduced.

That would make it easier for us to keep merging changes from upstream.

sandrine-bailleux-arm, 2017-02-06:
Hi @vwadekar,

As Dan said, we already run Coverity on TF but we're using the free, online version.
Unfortunately, this version has a limited set of features compared to the licensed one.
In particular, we don't seem to have any control over the analysis settings. For example,
I don't think it is possible to choose the set of checkers to run on the code.

Looking at Coverity's documentation on the Internet, it looks like one needs to pass some
options to the cov-analyze tool to run a MISRA analysis. If that's correct then I suppose
we can't get MISRA defects out of Coverity Scan Online (since we can't control the analysis
settings). Even if we could change the analysis settings, I'm not even sure this version
includes support for MISRA. I haven't found any reference in the sparse documentation
provided for Coverity Scan Online.

ARM also has a license for the full version of Coverity. We've been thinking about integrating
it in our development flow for some time but this requires some non-trivial changes in our
automated testing infrastructure, that we haven't had time to do yet. This work is not scheduled
at the moment.

I've seen that Coverity currently supports several versions of MISRA (MISRA-C 2004,
MISRA-C++ 2008, and MISRA-C 2012). I guess you'd be interested in compliance with
the latest version (2012)?

Regards,
Sandrine

vwadekar, 2017-02-06:
@sandrine-bailleux-arm 

Thanks. We would be interested in MISRA-C 2012. We do use the full version of Coverity and see a lot of HIGH impact issues in the common ATF code, hence want to clean it up.

Being MISRA clean is one of the mandatory requirements for our code releases. So, please let me know how we can help.

davidcunado-arm, 2017-02-09:
@vwadekar could you send the details from your Coverity analysis on the ATF code? That would give us an idea of the size of the work needed and so help with scheduling and prioritisation.

Email: david.cunado@arm.com

vwadekar, 2017-02-14:
@danh-arm

Thanks. I will send the details this week.

danh-arm, 2017-02-22:
We had a bit more of a look into MISRA compliance and the Coverity MISRA checker. MISRA rules are categorised in 7 levels with increasingly strict subsets of rules and directives within the standard. To comply with all rules at all levels would be very challenging. After running a scan internally and looking at the results from a high level, it seems the rules are broadly one of:

*   Rules that potentially add value.
*   Rules that are nit-picking, but are not difficult to apply.
*   Rules that are difficult/impossible to workaround (and may have unclear rationale).
*   Rules that conflict with our current coding standards or are otherwise controversial.

Also, I'm concerned a lot of these issues are in imported code (e.g. stdlib, libfdt), which might take a long time to fix and should be fixed in the upstream anyway. I guess we could look for alternative MISRA-compliant implementations but that requires further investigation.

In summary, I think this is going to be a "journey", whereby we fix all issues corresponding to a particular MISRA rule, on a rule-by-rule basis. We can't commit to any level of compliance at this stage, though.

From our side, the next steps will be to:

*   Regularly run a Coverity MISRA scan.
*   Starting from level 1 and working upwards, look at the results of the scan and categorise each rule Once we've got a better handle on the details, we can make a more specific plan for compliance.
*   For rules with clear value, raise and fix issues identified by the scan.

I'll leave this issue open as an "Enhancement" for tracking purposes. For non-controversial MISRA fixes, we're happy to take contributions at any time.

vwadekar, 2017-02-22:
@danh-arm

Thanks. I agree we have to work towards the goal at a steady pace. But
another important policy change that I would request is, to not allow
changes that add new MISRA defects. That way we can keep the baseline
constant and we can even take upstream changes without having to worry
about additional defects creeping in.

danh-arm, 2017-02-22:
> ... to not allow changes that add new MISRA defects

That's a reasonable goal, assuming the defect corresponds to an agreed "rule". We can investigate if the tools help track deltas against a baseline.

vwadekar, 2017-02-22:
@danh-arm 

Can you post the rules you want to enable first? Ideally, we should enable all, but we can "negotiate".

Even though we decide to fix only a subset of the rules, we should still enable all the rules for the daily scan and not allow new defects to creep in.

Suggestion: Enable all the rules which enable "Mandatory" and "Required" defects.

Thoughts?

danh-arm, 2017-02-23:
Hi @vwadekar 

> Can you post the rules you want to enable first?

We can enable them all, but fixing them is another matter. The MISRA rules will need reviewing in detail, which will take some time. We can't commit to this yet but we'll post something here when we have a plan.

> Even though we decide to fix only a subset of the rules, we should still enable all the rules for the daily scan and not allow new defects to creep in.

I agree, but preventing new defects can only happen once a fix plan is in place.

> Suggestion: Enable all the rules which enable "Mandatory" and "Required" defects.

As I said we can enable them all. I agree we should fix defects in the order "Mandatory"->"Required"->"Advisory". We would probably also prioritize BL31 fixes over other code. The Coverity scan levels may also allow sub-priotization.

antonio-nino-diaz-arm, 2018-12-19:
For more information regarding our efforts regarding MISRA check https://github.com/ARM-software/arm-trusted-firmware/wiki/ARM-Trusted-Firmware-Coverity-MISRA-Analysis

closed 2019-01-28 by antonio-nino-diaz-arm