ARM-software/tf-issues #456: Check if data cache is enabled in psci_do_pwrup_cache_maintenance()

vwadekar, 2017-02-06:
psci_do_pwrup_cache_maintenance() currently cleans stack memory and enables data cache. It relies on platforms not overriding the weakly defined bl31_plat_enable_mmu() function.

Platforms can override bl31_plat_enable_mmu() and decide to enable data cache along with the MMU. In this case, psci_do_pwrup_cache_maintenance() must check if data cache is already enabled and return early.

jeenu-arm, 2017-02-07:
If you're hinting at this call flow,

* `bl31_plat_enable_mmu`
* `psci_warmboot_entrypoint` -> `...` -> `psci_cpu_(suspend|on)_finish()` -> `psci_do_pwrup_cache_maintenance`

then, [PSCI integration guide](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/psci-lib-integration-guide.md) documents that:

> The Data cache must be enabled prior to invoking any of the PSCI library interfaces except for psci_warmboot_entrypoint().

Which would mean that `bl31_plat_enable_mmu` is not to enable data caches in this code path. The `flags` parameter to `bl31_plat_enable_mmu` is currently used to signal it from doing that.

danh-arm, 2017-02-07:
Hi @vwadekar. We are working on support for platforms like you describe. This will solve the issue you raised here  (among other things). We will contact you privately about this.

vwadekar, 2017-02-08:
@danh-arm 

Thanks.

vwadekar, 2017-02-15:
@danh-arm @jeenu-arm 

I tried calling psci_do_pwrup_cache_maintenance() immediately after bl31_plat_enable_mmu() and dont see any issues on my Tegra210 board.

So my question is, why cant we enable data cache immediately after enabling MMU? Why do we have a window where the D$ is disabled but the MMU is enabled?

danh-arm, 2017-02-16:
Hi @vwadekar 

You might not see any issues on your board but that does not mean it's safe to do for all boards. On platforms with multiple CPU clusters connected via a Cache Coherent Interconnect (e.g. ARM's CCI-500), when powering up for the first CPU in a non-primary cluster, it is necessary to program the CCI to enter coherency *before* enabling the data caches. This happens as part of the platform specific `pwr_domain_on_finish()` operation. However, it's still possible to take advantage of the other benefits of enabling the MMU earlier (e.g. enabling the instruction cache). Hence the window between enabling the MMU and enabling the data cache.

As you know from our private discussions, we are looking at optimizing this code path for platforms such as yours.

Regards

Dan.

danh-arm, 2017-02-22:
I don't think there is anything further to discuss here - closing.

vwadekar, 2017-02-22:
@danh-arm 

I think we still need to come up with a policy that work for Tegra210 as well.

danh-arm, 2017-02-23:
> I think we still need to come up with a policy that work for Tegra210 as well.

Yes, that's the what the code we've contacted you privately about is supposed to enable. That fixes the issue more elegantly that the solution you suggest here. If that is not working for you, then please continue the discussion in that thread.

vwadekar, 2017-03-02:
@danh-arm 

Hi Dan,

I believe the CPUs come up in a coherent state but the problem we see on Tegra210 is due to the stack not being in a clean state (?). As you might have noted, if we flush the stack before using it, we dont see the crashes.

vwadekar, 2017-03-08:
For Tegra SoCs, in order to enable D-cache earlier, I added the following code to tegra_bl31_setup.c

+/*******************************************************************************
+ * Handler to enable the MMU and data cache
+ ******************************************************************************/
+void bl31_plat_enable_mmu(uint32_t flags)
+{
+       /* Ensure any inflight stack writes have made it to main memory */
+       dmbst();
+
+       /* invalidate the stack memory for this CPU */
+       inv_dcache_range(plat_get_my_stack(), PLATFORM_STACK_SIZE);
+
+       /* enable MMU and the data cache */
+       enable_mmu_el3(0);
+}

Local testing looks promising.

Thoughts?

soby-mathew, 2017-03-13:
Hi Varun,
Just to confirm, you are referring to the case when [HW_ASSISTED_COHERENCY](https://github.com/ARM-software/arm-trusted-firmware/pull/860/commits/b0408e87f7dfbdfe3e00cd3c1421b2939dd209ca) is enabled for Tegra, right?

Assuming you have HW_ASSISTED_COHERENCY enabled for Tegra, yes, you need to invalidate the used stack before you turn on D-cache (but not the entire stack). The `psci_do_pwrup_cache_maintenance` enables the D-cache in addition to stack invalidation. Could you try the below sequence and let us know whether it works fine?

```
void bl31_plat_enable_mmu(uint32_t flags)
{
        /* enable MMU but leave D-cache disabled */
       enable_mmu_el3(DISABLE_DCACHE);

       /* Invalidate the used stack and enable D-cache */
       psci_do_pwrup_cache_maintenance();
}
```
If that resolves your issue, I guess we need to have a utility function in TF to do used stack invalidation which then can be reused by platforms which have HW_ASSISTED_COHERENCY.  Perhaps we could define a function `void do_used_stack_inv(void)` in misc_helpers.S 

```
func do_used_stack_inv
	stp	x29, x30, [sp,#-16]!

	/* ---------------------------------------------
	 * Ensure any inflight stack writes have made it
	 * to main memory.
	 * ---------------------------------------------
	 */
	dmb	st

	/* ---------------------------------------------
	 * Calculate and store the size of the used
	 * stack memory in x1. Calculate and store the
	 * stack base address in x0.
	 * ---------------------------------------------
	 */
	bl	plat_get_my_stack
	mov	x1, sp
	sub	x1, x0, x1
	mov	x0, sp
	bl	inv_dcache_range
        ldp	x29, x30, [sp], #16
	ret
endfunc do_used_stack_inv

```

This function can be reused by both platforms which have HW_ASSISTED_COHERENCY enabled and psci_do_pwrup_cache_maintenance().

vwadekar, 2017-03-13:
@soby-mathew 

Thanks. I like the idea of a helper function for invalidating the used stack. The code you mentioned works for Tegra, so you can push a change.

Can we allow platforms to decide when they want to call the new helper function? This way, we dont have to bind it to HW_ASSISTED_COHERENCY. Tegra, for one, would benefit from this approach as we would like to call the helper function even with HW_ASSISTED_COHERENCY=0

soby-mathew, 2017-03-14:
> Can we allow platforms to decide when they want to call the new helper function? This way, we dont have to bind it to HW_ASSISTED_COHERENCY. Tegra, for one, would benefit from this approach as we would like to call the helper function even with HW_ASSISTED_COHERENCY=0

Hi Varun,
No, it has be bound to HW_ASSISTED_COHERENCY because we make assumption about the platform based on this flag in PSCI Generic code. 

For example, see this code fragment [here](https://github.com/ARM-software/arm-trusted-firmware/pull/860/commits/b0408e87f7dfbdfe3e00cd3c1421b2939dd209ca#diff-a781c9e794f2bbae14f5453ad0f9a016R168) which does stack invalidation during power up.  To do the stack invalidation early in `bl31_plat_enable_mmu` and again in psci warm boot code with caches enabled would be incorrect. 

Maybe it is better to define a helper to enable MMU, D-Cache and the required stack inv the right way (`enable_mmu_with_stack_maint()` ?) since it is easy for platforms to make a mistake here. Need more thoughts on this.

vwadekar, 2017-03-15:
@soby-mathew 

I just want to separate the code to enable MMU, D-cache and required stack inv, from the HW_ASSISTED_COHERENCY, as it is helpful for Tegra210 even when HW_ASSISTED_COHERENCY=0

soby-mathew, 2017-03-17:
Hi Varun
As I mentioned in my previous comment, if you enable D-Cache early with HW_ASSISTED_COHERENCY=0, you are going to do stack invalidation [(here)](https://github.com/ARM-software/arm-trusted-firmware/pull/860/commits/b0408e87f7dfbdfe3e00cd3c1421b2939dd209ca#diff-a781c9e794f2bbae14f5453ad0f9a016R168) with D-Cache enabled which is functionally wrong. We dont want to enable a functionally incorrect behaviour.

vwadekar, 2017-03-20:
@soby-mathew 

Soby, the helper invalidate functions suggested earlier give platforms flexibility. We can always add a check in PSCI enable d-cache routine to see if d-cache is already enabled.

soby-mathew, 2017-03-20:
@vwadekar, We would like to avoid too many variation of configs executing the PSCI warmboot code. The PSCI warm boot sequence is already a complex scenario and enabling the D-cache early with HW_ASSISTED_COHERENCY=0 entitles us to cater for this additional case  which is a maintenance hazard IMO.  Hence I am not inclined not to give this flexibility.

vwadekar, 2017-03-20:
@soby-mathew 

The check in PSCI powerup cache maintenance would be a runtime check. Plus, the helper function that you suggested (do_used_stack_inv) can be used by Tegra only in the platform's enable_mmu handler. The common ARM code does not need any configs if we take this path.

soby-mathew, 2017-03-20:
The `config` which I was referring to was the state of the CPU when executing a code path (In this case,  with D-cache enabled early). We make assumptions about CPU state at various points in the warm boot sequence, this change will add another combination of assumptions that we have to cater for.

>the helper function that you suggested (do_used_stack_inv) can be used by Tegra only in the platform's enable_mmu handler

It would be used by all platforms with HW_ASSISTED_COHERENCY= 1

Also, it seems like you need some bits of HW_ASSISTED_COHERENCY (like enabling D-cache early) but not other bits. We need more convincing why this is the case and why this is not working on Tegra. Perhaps we need to continue this conversation privately as Dan suggested to understand better and maybe derive a generic solution.

vwadekar, 2017-03-20:
The only problem on Tegra with enabling D-cache early is the flush/inv of
the used stack memory. If flush/inv the used stack memory before enabling
the MMU/D-cache, everything works well.

vwadekar, 2017-03-20:
We need to enable D-cache early even for the HW_ASSISTED_COHERENCY=0 case.
This is to mitigate a hardware problem on Tegra210. We cannot execute from
TZSRAM memory, unless we enable D-cache early due to this problem.

soby-mathew, 2017-03-23:
@vwadekar, OK. In that case I would like to understand this problem.  I will email you privately regarding this.

soby-mathew, 2017-03-24:
Just to update here the outcome of the discussion: There are valid cases wherein D-Cache can be enabled early in warm boot path (eg: single cluster platforms) because there is no interconnect programming to done later.  This also helps Tegra 210 with their hardware problem.

soby-mathew, 2017-03-24:
Internal ref : https://jira.arm.com/browse/GENFW-1803

vwadekar, 2017-04-07:
@soby-mathew 

Any update on the solution?

soby-mathew, 2017-04-10:
@vwadekar, not yet. We hope to resolve it during this month.

vwadekar, 2017-04-10:
@soby-mathew 

Thanks for the update.

For now, we have merged the following patch in our downstream branch

```
diff --git a/plat/nvidia/tegra/common/aarch64/tegra_helpers.S b/plat/nvidia/tegra/common/aarch64/tegra_helpers.S
index 4eabea4..d03d778 100644
--- a/plat/nvidia/tegra/common/aarch64/tegra_helpers.S
+++ b/plat/nvidia/tegra/common/aarch64/tegra_helpers.S
@@ -1,5 +1,5 @@
 /*
- * Copyright (c) 2015, ARM Limited and Contributors. All rights reserved.
+ * Copyright (c) 2015-2017, ARM Limited and Contributors. All rights reserved.
  *
  * Redistribution and use in source and binary forms, with or without
  * modification, are permitted provided that the following conditions are met:
@@ -53,6 +53,7 @@
                                         ACTLR_EL3_CPUACTLR_BIT)
 
        /* Global functions */
+       .globl  plat_inv_used_stack_memory
        .globl  plat_is_my_cpu_primary
        .globl  plat_my_core_pos
        .globl  plat_get_my_entrypoint
@@ -154,6 +155,39 @@
 .endm
 
        /* -----------------------------------------------------
+        * void plat_inv_used_stack_memory(void);
+        *
+        * This function invalidates the used stack memory for
+        * the current CPU
+        * -----------------------------------------------------
+        */
+func plat_inv_used_stack_memory
+       stp     x29, x30, [sp,#-16]!
+
+       /* ---------------------------------------------
+        * Ensure any inflight stack writes have made it
+        * to main memory.
+        * ---------------------------------------------
+        */
+       dmb     st
+
+       /* ---------------------------------------------
+        * Calculate and store the size of the used
+        * stack memory in x1. Calculate and store the
+        * stack base address in x0.
+        * ---------------------------------------------
+        */
+       bl      plat_get_my_stack
+       mov     x1, sp
+       sub     x1, x0, x1
+       mov     x0, sp
+       bl      inv_dcache_range
+
+       ldp     x29, x30, [sp], #16
+       ret
+endfunc plat_inv_used_stack_memory
+
+       /* -----------------------------------------------------
         * unsigned int plat_is_my_cpu_primary(void);
         *
         * This function checks if this is the Primary CPU
diff --git a/plat/nvidia/tegra/common/tegra_bl31_setup.c b/plat/nvidia/tegra/common/tegra_bl31_setup.c
index 1d04a6c..0a505c9 100644
--- a/plat/nvidia/tegra/common/tegra_bl31_setup.c
+++ b/plat/nvidia/tegra/common/tegra_bl31_setup.c
@@ -1,5 +1,5 @@
 /*
- * Copyright (c) 2015-2016, ARM Limited and Contributors. All rights reserved.
+ * Copyright (c) 2015-2017, ARM Limited and Contributors. All rights reserved.
  *
  * Redistribution and use in source and binary forms, with or without
  * modification, are permitted provided that the following conditions are met:
@@ -399,3 +399,20 @@ int bl31_check_ns_address(uint64_t base, uint64_t size_in_bytes)
        /* valid NS address */
        return 0;
 }
+
+/*******************************************************************************
+ * Handler to enable the MMU and data cache
+ ******************************************************************************/
+void bl31_plat_enable_mmu(uint32_t flags)
+{
+       (void)flags;
+
+       /* invalidate the stack memory for this CPU */
+       plat_inv_used_stack_memory();
+
+       /* enable data cache */
+       write_sctlr_el1(read_sctlr_el1() | (1UL << 2));
+
+       /* enable MMU */
+       enable_mmu_el3(0);
+}
diff --git a/plat/nvidia/tegra/include/tegra_private.h b/plat/nvidia/tegra/include/tegra_private.h
index 4ddadb7..75a06fe 100644
--- a/plat/nvidia/tegra/include/tegra_private.h
+++ b/plat/nvidia/tegra/include/tegra_private.h
@@ -74,6 +74,9 @@ typedef struct irq_sec_cfg {
        uint32_t type;
 } irq_sec_cfg_t;
 
+/* Declarations for tegra_helpers.S */
+void plat_inv_used_stack_memory(void);
+
 /* Declarations for plat_psci_handlers.c */
 int32_t tegra_soc_validate_power_state(unsigned int power_state,
                psci_power_state_t *req_state);
diff --git a/services/std_svc/psci/psci_helpers.S b/services/std_svc/psci/psci_helpers.S
index 6ccf943..ca09c07 100644
--- a/services/std_svc/psci/psci_helpers.S
+++ b/services/std_svc/psci/psci_helpers.S
@@ -122,6 +122,15 @@ func psci_do_pwrup_cache_maintenance
        stp     x29, x30, [sp,#-16]!
 
        /* ---------------------------------------------
+        * Check if data cache is already enabled by the
+        * platform code..
+        * ---------------------------------------------
+        */
+       mrs     x0, sctlr_el3
+       and     x0, x0, #SCTLR_C_BIT
+       cbnz    x0, 1f
+
+       /* ---------------------------------------------
         * Ensure any inflight stack writes have made it
         * to main memory.
         * ---------------------------------------------
@@ -149,6 +158,6 @@ func psci_do_pwrup_cache_maintenance
        msr     sctlr_el3, x0
        isb
 
-       ldp     x29, x30, [sp], #16
+1:     ldp     x29, x30, [sp], #16
        ret
 endfunc psci_do_pwrup_cache_maintenance
```

soby-mathew, 2017-04-12:
@vwadekar, Thanks for that. We have a patch which is undergoing internal review currently and will notify you once we have the PR.

soby-mathew, 2017-04-13:
@vwadekar , could you try the PR and see if it works for you ? You need to enable the new build option WARMBOOT_ENABLE_DCACHE_EARLY.

closed 2017-04-21 by davidcunado-arm