ARM-software/tf-issues #458: Import mbed TLS

masahir0y, 2017-02-07:
Hi. 

If I see under drivers/auth/, mbed TLS is the only supported library for Trusted Board Boot in upstream.
I tested it, and it worked on my board.  One thing I do not like is, it is not user-friendly to download mbed TLS separately, then give command line option `MBEDTLS_DIR=<path-to-mbedtls>` when building.

I'd like to suggest to include necessary library files into this project.
Is there any problem to do so?

The mbed TLS is licensed under Apache license, so I am guessing there is no license problem.
For maintainability, I think it is not much burden to sync files periodically.  Actually, we are doing it for lib/libfdt already.

Thought?

danh-arm, 2017-02-14:
> Is there any problem to do so?

There are no license issues but there are commercial reasons why we can't do this. I do see the current process is a bit non-user-friendly, though. One possible compromise is to make mbed TLS a git submodule of TF, which would simplify installation. There may be other similar solutions.

sbranden, 2017-02-14:
Hi Dan,

Using git submodule is fine so long as there is a mechanism to choose version of mbedtls in use?  Currently we created a yocto build recipe to handle building atf + checkout mbedtls.

Scott

masahir0y, 2017-02-28:
> There are no license issues but there are commercial reasons why we can't do this.

I am a bit disappointed with this because it sounds like
this project is not a healthy open source project.
It is under the strong influence of a particular company instead of the movement of an open community.