ARM-software/tf-issues #459: Trustzone and asynchronous external abort from NW

vsiles, 2017-02-15:
Hi !
I'm designing an armv7 secure monitor, and I'm facing some situation where the normal world (NW) generates asynchronous external aborts that triggers during the monitor/secure world, because they happened too close from a world switch. I configured the SCR.EA / AW bits so that NW can't mask them and they land in the secure monitor, but even with this configuration, I'm still getting some asynchronous abort from the NW into the SW.

My current code is performing a dsb;isb sequence before checking the ISR register in the monitor, to try to "flush" the external abort before scheduling the SW.

I'm wondering how the ATF deals with such situation ?

Best,
V.

danh-arm, 2017-02-15:
Hi @vsiles 

Trusted Firmware currently doesn't do anything sophisticated here. It just enables the `SCR.EA` bit early and by default clears it before ERET'ing to the normal world, so that all asynchronous EAs (whatever their source) end up being handled in the normal world.

I think any any attempt to switch the value of `SCR.EA` as part of the monitor's world switch is doomed to failure, since you can never guarantee that the asynchronous EA will be handled by the same EL that generated it. Barriers won't help here. This is discussed (among other things) in #368, in particular in #368#issuecomment-196329027.

A better solution is for EL3 to always initially handle the EA (whatever the source) and then allow the normal world to register to handle these events. This is captured in #374, which is something we intend to address this year.

Regards

Dan.

vsiles, 2017-02-15:
Thank you @danh-arm , I think we are going to go for the 'mask them by default and let some configuration option decide if NW or SW wants to deal with them' solution, which allows us to crash linux during demo session without loosing the SW ;)

Since #368 is quite well documented, I'll close this question.
Best,
V.

closed 2017-02-15 by vsiles