ARM-software/tf-issues #461: utils.h causes compile issue with building host tool

sbranden, 2017-03-08:
The following commit causes compile issues when utils.h is included building host tools that we have developed:

https://github.com/ARM-software/arm-trusted-firmware/commit/308d359b260d888f024a2d26c76cd4a50789e432

Above change to utils.h #include <type.h>

Could you move the newly added functionality and #include of type.h to a different header and leave utils.h alone?

douglas-raillard-arm, 2017-03-14:
Hi @sbranden,

utils.h and more generally the headers in the include/lib are meant to be used for TF binaries. We do not support inclusion of them in a radically different environment than target binaries (BLx) and platform code. Given the content of this specific util.h header, I suggest taking the needed pieces out in a separate header for your out-of-tree tool.

sbranden, 2017-03-14:
The tools we are building are for such items as generating static MMU tables for ATF builds.  There are not for a radically different environment - they are for host tools for our use with ATF.

I suggest we split the defines in utils.h in one file and the newly added functioning prototypes which include type.h in another.  This will solve all the issues we have.  Or if you like, I can put the defines in one header and have utils.h include that header and leave the new prototypes and #include of type.h in utils.h

sandrine-bailleux-arm, 2017-03-15:
Hi @sbranden 

> The tools we are building are for such items as generating static MMU tables for ATF builds.

This tool sounds interesting! So I guess the tool generates some C file that contains the translation
table array and its initialisation values? Something like
```
uint64_t xlat_tables[] = {
  0x...,
  0x...,
  /* ...and so on... */
} __section("xlat_table");
```
And then you include this generated C file in the build. Is that the idea?

And I guess that in this context, you don't need to use the translation table library (provided under
the lib/xlat_tables/ directory) because you don't need to populate any page tables at run time?

Would you be able to share the code source of this tool?

Regards,
Sandrine

sbranden, 2017-03-15:
Hi Sandrine,

The tool took quite some time to develop.  With it we now place the BL1 and BL2 tables in ROM rather than wasting time and RAM generating them on the fly.

Change involves:
- adding mmu_table_create host tool to generate mmu tables that can be compiled into image
- adding section to linker file. 
- adding compile options into xlat_tables code to replace init_xlation_table and disable most of the other functions.

I don't think this fits into your ATF codebase though.  According to Douglas-Raillard this is radically different than what your ATF codebase is supposed to be used for.  You don't support such inclusion and it looks like we should continue to support this out of tree?

dp-arm, 2017-03-17:
Hi @sbranden 

If you could make the tool available we could evaluate it for use in ARM TF.  You may have misinterpreted Douglas's argument as I don't think it is particularly relevant in this case.  We already have host tools in TF (cert_create and fiptool) so having another host tool is not beyond the realm of possibility, if indeed there is a good use for it.

sbranden, 2017-03-17:
I'll try to extract the changes once I upgrade to the latest ATF.  In order to do so I need utils.h fixed.  So, can you move types.h out of utils.h or move all the defines in utils.h somewhere else (utils_def.h?) and have utils.h include that header?

dp-arm, 2017-03-21:
Why can you not just copy utils.h and adapt as needed?

sbranden, 2017-03-22:
@dp-arm, the tool includes platform_def.h.  And platform_def.h needs defines in utils.h.  So, by placing an include of a custom types.h in utils.h there is more the definitions in utils.h now.  We need a utils_def.h that contains only defs to solve this problem.

closed 2017-05-02 by davidcunado-arm