ARM-software/tf-issues #466: Can a NS-EL1 interrupt take over the processor executing in BL31

chinnabha, 2017-04-01:
Does the current ATF routing model allows the processor executing in BL31 (EL3) to jump to NS-EL1 exception level to handle an incoming NS-EL1 interrupt? I am not sure whether BL31 is fully atomic or specific portions of it are atomic.

If the answer to above question is yes: I have some BL31 code where in it is waiting on interrupt (via WFI). A NS-EL1 interrupt got triggered in that scenario. Can the processor in this scenario wake up to jump to the corresponding NS-EL1 interrupt vector table to service this interrupt?

soby-mathew, 2017-04-03:
If I understand correctly, there are 2 scenarios being referred to in your question. 

The first one is whether BL31 is pre-emptible to allow handling NS-EL1 interrupts. The simple answer to that is no but if you referring to a case wherein the S-EL1 is handling a `yeilding SMC` then the S-EL1 can be pre-empted and BL31 allows returning back to normal world to handle the interrupt. Note that BL31 is still non-preemptible and S-EL1 is the one pre-empted in this case.

The second case is whether the Non Secure Interrupt can wake up a CPU from suspend. Yes, it can but it depends on whether the power controller can respond to the interrupt and wake-up the CPU. 

> I have some BL31 code where in it is waiting on interrupt (via WFI)

The BL31 upstream code only has WFI for power-down/suspend/retention. If you are referring to that, then it falls in the scenario-2 discussed above. Otherwise, BL31 is non-preemptible.

chinnabha, 2017-04-03:
Thanks @soby-mathew. The scenario of interest is case 2, specifically the power down scenario. But the case 2 seems contradicts the fact that BL31 is non-preemptible.  How the processor in BL31 (EL3) suspend can wake and go to NS-EL1? Even though the power controller or the gic responds to the incoming NS-EL1 interrupt and wakes the processor, can the the non-preemptible nature of BL31 allows the processor to go to NS-EL1 world ?

soby-mathew, 2017-04-03:
>  How the processor in BL31 (EL3) suspend can wake and go to NS-EL1?
The BL31 will `return` back to the Non secure world (as in `return` from a function call) by executing warm_boot sequence on wake-up from powerdown as specified by PSCI. I may not be understanding your question but this does not have anything to do with preemption.

chinnabha, 2017-04-04:
Sorry, I didn't give you the scenario. On my board, the primary core is waking up after wfi() and the panic is seen. The only source I can think of for this is NS-EL1 scheduler timer interrupts wakeup. What may be the expected implementation from the system controller processor (SCP) on the PLLs (turn OFF ??) that supplies power to GIC and NS-EL1 scheduler timers in scpi_sys_power_state()?  

What might be the reason to not to disable the gic cpu interface in power down? Why ATF doesn't disable gic cpu interface after successful scpi_sys_power_state()? 

void __dead2 css_system_off(void)
{
        uint32_t response;

        /* Send the power down request to the SCP */
        response = scpi_sys_power_state(scpi_system_shutdown);

        if (response != SCP_OK) {
                ERROR("CSS System Off: SCP error %u.\n", response);
                panic();
        }
        wfi();
        ERROR("CSS System Off: operation not handled.\n");
        panic();
}

soby-mathew, 2017-04-06:
@chinnabha, As mentioned in the PSCI specification section 4.1 [1],  wakeup from powerdown means "A reset of the core, after the power has been restored. So if the CPU wakes up  due to a NS-EL1 interrupt, then the core will start from reset and execute the `warmreset` path. 

Hence, if the CPU is not doing a reset on wake-up from power down WFI, then the power controller is not doing something right on your platform.

> What may be the expected implementation from the system controller processor (SCP) on the PLLs (turn OFF ??) that supplies power to GIC and NS-EL1 scheduler timers in scpi_sys_power_state()? Are those relevant here?

Depends on the level of power down and it is platform specific. For a CPU level powerdown, the per-cpu timers and the CPU loses context but the GIC usually retains power. For a system level power down, even the GIC loses can lose power and the GIC should be restored by software. 

So it is a  platform decision as to which peripherals lose power at which level.

> What might be the reason to not to disable the gic cpu interface in power down? Why ATF doesn't disable gic cpu interface after successful scpi_sys_power_state()?  void __dead2 css_system_off(void)

ARM TF does disable GIC cpu interface for CPU suspend (see [here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/plat/arm/css/common/css_pm.c#L133)). The code you have referenced is for system shutdown which shuts down the entire system and a full system reset is expected when restarted later.

[1] http://infocenter.arm.com/help/topic/com.arm.doc.den0022c/DEN0022C_Power_State_Coordination_Interface.pdf

chinnabha, 2017-04-17:
Thanks @soby-mathew for the help.

closed 2017-04-17 by chinnabha