ARM-software/tf-issues #467: How can i disable all the console output?

prime-zeng, 2017-04-03:


antonio-nino-diaz-arm, 2017-04-03:
Building the firmware with `LOG_LEVEL=0` and `DEBUG=0` should be enough. Unfortunately, some of the calls to tf_printf() aren't disabled with `LOG_LEVEL=0` and it is needed to set `DEBUG=0` as well.

prime-zeng, 2017-04-05:
So the LOG_LEVEL = 0 and DEBUG = 0 can disable the call to tf_printf, what about the printf?  I can't find anycode which affects printf?

antonio-nino-diaz-arm, 2017-04-05:
`printf()` isn't used by the firmware itself, just by the tools under the `./tools/` folder.

Well, actually, there's only one place where it is used in the firmware, and that is `plat/rockchip/rk3399/drivers/dram/dfs.c:448`. It should probably be changed to `tf_printf`.

jwerner-chromium, 2017-04-05:
Whoops, yeah, looks like we're actually linking two completely separate printf implementations into our BL31 there because of that. That's not good at all. @Caesar-github, would you mind uploading a fix for that (here and in Chromium firmware branch)?

Antonio, does the code only meant for external tools really need to be under the same directory structure as normal firmware code (under lib/*)? Seems like it would be better to pull that apart more explicitly (e.g. something like tools/common/) so that a typo like this would fail obviously and not just silently link all this stuff in. It's also not at all clear to me what other library code is not intended to be used in firmware platform code (e.g. all of lib/stdlib/ or only printf?).

rkchrome, 2017-04-06:
Julius:
Okay, posted on https://github.com/ARM-software/arm-trusted-firmware/pull/892

prime-zeng, 2017-04-06:
@rkchrome   I think tf_printf should not be directly called, NOTICE/INFO should be used instead,  tf_printf will unconditionally print.

rkchrome, 2017-04-06:
Indeed, I will change the tf_printf for now.

rkchrome, 2017-04-06:
Updated it for rockchip platform.

Look like,  tf_printf need to change under /lib/ and the drivers/arm/cci/cci.c

drivers/arm/cci/cci.c
57:			tf_printf("Slave interface ID is invalid\n");
62:			tf_printf("Multiple masters are assigned same"
70:		tf_printf("No master is assigned a valid slave interface\n");

antonio-nino-diaz-arm, 2017-04-06:
> Antonio, does the code only meant for external tools really need to be under the same directory structure as normal firmware code (under lib/*)? Seems like it would be better to pull that apart more [...]

I'm not sure I understand what you say, @jwerner-chromium. Tools in the tool folder don't use anything in the lib folder. If you take a look at, for example, `tools_cert/create/Makefile`, it doesn't use the files under lib/stdlib, that's only used by the firmware, tools use whatever version of stdlib the compiler wants to link them to.

Does this answer your questions?

PS: Well, that's not exactly true, the makefiles do copy some files from other folders. To be precise, cert_create copies files `include/lib/stdlib/sys/uuid.h` and `include/common/firmware_image_package.h` into its own folder before compiling anygthing.

antonio-nino-diaz-arm, 2017-04-06:
@rkchrome At least those tf_printf are removed in release builds. But they should change as well and use ERROR, I agree.

jwerner-chromium, 2017-04-06:
Sorry, @antonio-nino-diaz-arm, then I guess I misunderstood this comment from you above:

> printf() isn't used by the firmware itself, just by the tools under the ./tools/ folder.

So if lib/stdlib/printf.c isn't used by tools and isn't (supposed to be) used by the firmware, then what is it used for? Why is it there at all?

My main question/suggestion is that if tf_printf() is the only printf implementation that's supposed to be used in firmware, there shouldn't be a second printf() in the common library that is compiled by default and part of the normal include path. It is very easy to accidentally write a line with printf() instead of tf_printf(), and when that happens it should lead to a compiler/linker error (so that people will notice), it shouldn't silently pull in code for a second printf() implementation that you weren't supposed to use.

soby-mathew, 2017-04-06:
@jwerner-chromium, I can see your point. The printf() includes a whole family of related functions like snprintf, vsnprintf() which are sometimes required for integrating external library components (like mbedTLS) in firmware.  Also, printf() supports lot more format strings than that are supported by tf_printf() and may prove useful for some cases.

IMO, I think firmware should have access to printf() (and related APIs) but it would also be good if we can flag its inadvertent use.

antonio-nino-diaz-arm, 2017-04-07:
Every function of that family uses the same function internally, `kvprintf`. Removing `printf` wouldn't help much as it is only a wrapper and the firmware still needs functions like `snprintf` (see `drivers/partition/partition.c:51` for example).

However, it may be worth converting `tf_printf` into a wrapper around a common function so that we can save memory by having `tf_printf` and something like `tf_snprintf` that use the same code internally. Then, we could define `printf` and `snprintf` as `tf_printf` and `tf_snprintf`, and maybe others if needed.

Alternatively, we could remove the current `tf_printf`, define `tf_printf` to `printf` and just remove the code we don't need from the `kvprintf` implementation in `lib/stdlib`.

sbranden, 2017-04-07:
Whatever printf improvements are made.  Please: do not increase the code size of BL2.

Also, please do not remove any existing functionality.  It may not be used in your ATF drivers or application but should be available for others to use.  Code size optimizations are welcome but link or compile time configuration options should be used to reduce size and still support existing functionality.

antonio-nino-diaz-arm, 2017-04-10:
Actually, BL2 size was my main concern. Right now, a quick look at `bl2.dump` tells me that, when compiling with `TRUSTED_BOARD_BOOT=1`, we have two functions that do the same thing and that are quite big: `kvprintf` (3424 bytes) and `tf_printf` (796 bytes).

Another option could be to compile out `tf_printf` if `kvprintf` is used and define `tf_printf` it as `printf`.

antonio-nino-diaz-arm, 2017-04-10:
I forgot about the other discussion around this. If we implement the changes suggested in https://github.com/ARM-software/arm-trusted-firmware/pull/883#issuecomment-291931113 , the `plat_printf` define could be simply pointed at `printf` (if the platform knows it's going to use the stdlib functions anyway) and get the same effect. The advantages are that we have less options to worry about in the generic code and it's more flexible.

danh-arm, 2017-07-06:
I think the original question is answered here. The other discussion can be continued elsewhere if needed (e.g. in https://github.com/ARM-software/tf-issues/issues/462)

closed 2017-07-06 by danh-arm