ARM-software/tf-issues #469: what is purpose of xlat_tables_v2?

sbranden, 2017-04-11:
Could you please provide information on the purpose of xlat_tables_v2.  If it takes up more code and data what are its benefits?  What is wrong with the original xlat_tables that is trying to be solved?

Also, could you please maintain support for the arm reference platforms back for original xlat_tables?  I had to revert the commit that changed the arm platforms to use xlat_tables_v2 in order for them to work with xlat_tables.

danh-arm, 2017-04-13:
The initial feature that xlat_tables_v2 provides is the ability to dynamically add and remove memory regions, even after the MMU is enabled. One use-case for this is to temporarily map memory into the secure world (see https://github.com/ARM-software/arm-trusted-firmware/pull/897 for an example).

This new version provides a more solid base for future use-cases, for example unmapping transient code at runtime or allocation of virtual addresses (VAs) to reduce the overall VA space. In time this should enable a reduction in the overall memory footprint, despite the additional library footprint. Until such time we'll keep the original version upstream but the intention is to deprecate it eventually.

> Also, could you please maintain support for the arm reference platforms back for original xlat_tables?

No, the ARM reference platform code is owned by ARM and we choose to use xlat_tables_v2. We don't provide any compatibility guarantees for other platforms that use the code in plat/arm, although we will try to keep any upstream platforms that use plat/arm code working (currently ZyncMP is the only such platform).

danh-arm, 2017-04-19:
Actually, regarding the ARM reference platforms, we've found it useful to support both versions for testing purposes. So a patch will be coming soon that allows choice of either (but with v2 as the default).

sbranden, 2017-04-19:
Thanks - that makes more sense rather than removing features that are used.

closed 2017-07-06 by danh-arm