ARM-software/tf-issues #470: Can we use another register replace with x18?

kevinzhang1986, 2017-04-13:
It's mentioned that we shouldn't use x18 in firmware code in ARM SBSA spec. Because the operation system will use x18 in Linux ABI, and maybe conflict with firmware. So Can we another register replace with x18 in trusted firmware.

danh-arm, 2017-04-13:
> It's mentioned that we shouldn't use x18 in firmware code in ARM SBSA spec.

I couldn't find such a mention in the SBSA spec - could you provide a more specific reference? Or do you actually mean the PCS (Procedure Call Standard)?

> So Can we another register replace with x18 in trusted firmware.

Are you asking for Trusted Firmware to avoid using x18 and use another callee saved register instead? This actually may be a good idea since the PCS says "It should not be assumed that treating the register [x18] as callee-saved will be sufficient to satisfy the requirements of the platform".

Is this a theoretical issue or are you seeing a real problem?

kevinzhang1986, 2017-04-14:
Sorry, My mistake. It's mentioned in UEFI Spec 2.6, chapter 2.3.6.4 Detailed Calling Convention. I think it's also suitable for TF.

danh-arm, 2017-04-19:
Thanks. For the benefit of others, here's the relevant extract:

> The EFI AArch64 platform binding defines the platform register (r18) as “do not use”. Avoiding
> use of r18 in firmware makes the code compatible with both a fixed role for r18 defined by the
> OS platform ABI and the use of r18 by the OS and its applications as a temporary register.

So this sounds like a sensible thing to do but I'd still like to understand if you are seeing a real problem in practice, as that will help define the priority.

kevinzhang1986, 2017-04-24:
We haven't see a real problem in practice, but we still hope TF can follow the spec, thanks :)

danh-arm, 2017-04-25:
Internal ref:
https://jira.arm.com/browse/GENFW-1853

danh-arm, 2017-04-26:
We had some more discussion on this (thanks to @athoelke).

The ARM PCS allows an OS ABI to define a role for x18 (e.g. "referencing static data tables" or just "free for use as a temporary"). UEFI has to operate in the same execution context as an OS and so needs to respect the OS use of x18, which it cannot know in advance. The only way to achieve this is to never modify the register.

ARM TF does not operate in the context of an OS; it effectively defines its own ABI context and so can make its own decision how to use x18. The EL3 entrypoints do the right thing by always saving and restoring the lower EL's x18 value (as required by the SMCCC), which gives TF the freedom to use it as a temporary register within EL3.

So it looks like there's nothing to do here.

closed 2017-07-06 by danh-arm