ARM-software/tf-issues #472: fiptool: fiptool remove syntax for blobs is non-intuitive

xypron, 2017-04-21:
Arm trusted firmware provides tool fiptool.

To remove a block by specifying a UUID the following syntax has to be used:

    fiptool remove --blob uuid=UUID,file= FILE

Having to specify ",file=" does not make any sense here.

The syntax should be changed to

    fiptool remove --blob uuid=UUID FILE

Best regards

Heinrich Schuchardt

dp-arm, 2017-04-24:
May I ask why you think it is necessary to specify file in the remove command?  I just ran this locally:

$ ./fiptool create --blob uuid=01234567-89ab-cdef-0123-456789abcdef,file=image.bin fip.bin
$ ./fiptool info fip.bin
01234567-89AB-CDEF-0123-456789ABCDEF: offset=0x60, size=0x1FAD0, cmdline="--blob"
$ ./fiptool remove --blob uuid=01234567-89ab-cdef-0123-456789abcdef fip.bin
$ ./fiptool info fip.bin
$ echo $?
0

danh-arm, 2017-04-25:
This appears to be a non-issue. Please re-open if otherwise.

closed 2017-04-25 by danh-arm