ARM-software/tf-issues #474: Minor Makefile Problems

EvanLloyd, 2017-05-03:
Building on a machine with 40 processors has revealed minor problems in the makefiles.
    1.  The macro MAKE_PREREQ_DIR has a minor bug, in that it is capable of
         generating recursive dependencies.
    2.  The inclusion of BUILD_DIR in TEMP_OBJ_DIRS left no explicit
        dependency, BUILD_DIR may not exist when subdirectories are
        created by a thread on another CPU.
    3.  Some recent changes have added direct use of the echo command without
        parameters.  This fails on a Windows shell, because echo without
        parameters reports the mode ("ECHO is on").

antonio-nino-diaz-arm, 2017-05-10:
This is fixed, isn't it? I suppose that GitHub hasn't closed it because the text of the PR https://github.com/ARM-software/arm-trusted-firmware/pull/926 doesn't say "Fixes X" but "Fixes for X", but I just want to confirm that everything in this issue has been fixed in the master branch.

danh-arm, 2017-05-15:
Yes, it's fixed - closing

closed 2017-05-15 by danh-arm