ARM-software/tf-issues #476: Using TSP as BL32 and using smc instruction back to BL31

rangewang, 2017-05-10:
I just study ATF few months ago and something confused me.

When BL31 discovers "bl32_ep_info" not a NULL, BL31 will jumps to function of "[tsp_entrypoint](https://github.com/ARM-software/arm-trusted-firmware/blob/f132b4a05b23916c1101add4bd6d973a99983719/bl32/tsp/aarch64/tsp_entrypoint.S#L49)" to initiate BL32(TSP). 

And I use TSP as BL32 and it can initiate as well, but at the end of "tsp_entrypoint" function.
It touch an instruction "[smc #0](https://github.com/ARM-software/arm-trusted-firmware/blob/f132b4a05b23916c1101add4bd6d973a99983719/bl32/tsp/aarch64/tsp_entrypoint.S#L146)" and something wrong happen......"panic_handler"

I know "smc" is ARMv8 specific and it can switch betweent normal world and secure world.(I read the document of "[SMC CALLING CONVENTIOTION](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.den0028b/index.html)") 

but until now, I still can't understand if using "smc #0" instruction what will happen？

jump to an exception vector and take control flow back to BL31？(not sure)

danh-arm, 2017-05-15:
Hi

The TSP (or any BL32) executes at Secure Exception Level 1 (S-EL1). The SMC instruction in BL32 generates an exception in the BL31 code executing at EL3. In the code you refer to, the function id `TSP_ENTRY_DONE` is passed in x0. In the AArch64 case, execution should resume in the `sync_exception_aarch64` vector in bl31/aarch64/runtime_exceptions.S`.

The TSP Dispatcher (or TSPD) is a component that is integrated into BL31. In this case, the code should eventually end up in the `TSP_ENTRY_DONE` case statement in `tspd_smc_handler()` (`services/spd/tspd/tspd_main.c`).

But it sounds like execution is not getting that far for you. You should check that BL31 is built with the `SPD = TSPD` option. Other than that, it's hard to guess what is going wrong.

I hope that gives you some ideas.

rangewang, 2017-05-16:
Hi danh-arm

Thanks for your response.
as you tell above, my BL31 is built with SPD=TSPD and code flow is like this 

from BL31 [(*bl32_init)](https://github.com/ARM-software/arm-trusted-firmware/blob/f67c66311e86177a3cd6206b47a15c71233f7c47/bl31/bl31_main.c#L101) jump to [func tsp_entrypoint](https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl32/tsp/aarch64/tsp_entrypoint.S#L49) (bl32/tsp/aarch64/tsp_entrypoint.S) and tsp_main function also initiate well. And the flow touch the "[smc #0](https://github.com/ARM-software/arm-trusted-firmware/blob/f132b4a05b23916c1101add4bd6d973a99983719/bl32/tsp/aarch64/tsp_entrypoint.S#L146)"

After the function id TSP_ENTRY_DONE stored in x0 and do the "smc #0" instruction, the flow will jump to [vector_entry sync_exception_aarch64](https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl31/aarch64/runtime_exceptions.S#L206) and enter "macro handle_sync_exception"

but some thing wrong with this line "[mrs	x30, esr_el3](https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl31/aarch64/runtime_exceptions.S#L41)". when it load the value from ESR_EL3 but the value of EC(Exception Class) field is 0x0, not the [#EC_AARCH64_SMC](https://github.com/ARM-software/arm-trusted-firmware/blob/f67c66311e86177a3cd6206b47a15c71233f7c47/include/lib/aarch64/arch.h#L399) 0x17

so I can not do well at continuous flow
[cmp	x30, #EC_AARCH64_SMC](https://github.com/ARM-software/arm-trusted-firmware/blob/f132b4a05b23916c1101add4bd6d973a99983719/bl31/aarch64/runtime_exceptions.S#L48) 
b.eq	smc_handler64

Therefore flow can not end up in the TSP_ENTRY_DONE.

danh-arm, 2017-05-16:
Hi

An EC of 0x0 means "Unknown reason". The ARM ARM (section D7.2.27) lists a number of possible reasons for why this might happen. It's hard for me to speculate what this could be but one thing to check could be whether SMC instructions are disabled in your platform by SCR_EL3.SMD (this is not the case normally). Also, have you checked if your platform implements all the architected exception levels?

rangewang, 2017-05-22:
Hi

I have checked the SMC instruction is enable by SCR_EL3.SMD and this bit in ATF default is enable.
In my platform, I implemented EL3(for BL31), EL1(BL32) and in secure world EL2 was not exit. (Now I didn't consider BL33, because it was not reach the state of flow)

And until now I can't understand why the EC value in ESR_EL3 still keeping original value when doing SMC instruction.

I also read the "Exceptions with an unknown reason"...

So, my thinking is after doing the SMC instruction and the ESR_EL3.EC will update automatically, right ?

rangewang, 2017-05-23:
Hi

Finally, I find that my  firmware version of In-Circuit Debugger had some bug with ESR_EL3 register. 
So all the ATF flow is correct, I just update my firmware with the latest version and ESR_EL3 work normally.

Thank you!

danh-arm, 2017-05-23:
I'm glad you fixed the problem! Closing this issue....

closed 2017-05-23 by danh-arm