ARM-software/tf-issues #485: Runninf PSCI performance test

vingu-linaro, 2017-05-31:
I'd like to run performance test that is described in the wiki page : https://github.com/ARM-software/arm-trusted-firmware/wiki/ARM-Trusted-Firmware-PSCI-Performance-on-Juno
I have recompiled my atf with ENABLE_RUNTIME_INSTRUMENTATION=1 but i don't know how to get the results.
Can you help me to get the results ?
Thanks

soby-mathew, 2017-05-31:
@vingu-linaro, The results are based on an ARM-internal test framework. Although the tests themselves are not upstream, the  timestamps collected as part of BL31_RUNTIME_INSTRUMENTATION results can be queried using the PMF_SMC_GET_TIMESTAMP_64 smc.  Ideally it is good to have the test code available upstream, but that is not the case currently.

dp-arm, 2017-06-01:
Just to expand a bit on Soby's reply.

The supported timestamps are the following (from runtime_instr.h):

```
#define RT_INSTR_ENTER_PSCI             0
#define RT_INSTR_EXIT_PSCI              1
#define RT_INSTR_ENTER_HW_LOW_PWR       2
#define RT_INSTR_EXIT_HW_LOW_PWR        3
#define RT_INSTR_ENTER_CFLUSH           4
#define RT_INSTR_EXIT_CFLUSH            5
```

A typical PSCI call might go through this sequence:

`ENTER_PSCI`, `ENTER_HW_LOW_PWR`, `EXIT_HW_LOW_PWR`, `EXIT_PSCI`.

If the L2/L3 caches need to be flushed, `ENTER_CFLUSH`/`EXIT_CFLUSH` timestamps will also be recorded.

To retrieve these timestamps, an SMC with 3 arguments need to be invoked.

The first argument should be `PMF_SMC_GET_TIMESTAMP_64`, the second argument should be the `tid`.  To construct the `tid`, you can do something like:

```
                tid = PMF_ARM_TIF_IMPL_ID << PMF_IMPL_ID_SHIFT;
                tid |= PMF_RT_INSTR_SVC_ID << PMF_SVC_ID_SHIFT | RT_INSTR_ENTER_PSCI;
```

This can be done in a loop for all supported timestamps.

Once that's done, it is a good idea to have a condition to check the values of the timestamps to ensure they make sense (i.e. `EXIT_PSCI` cannot be less than `ENTER_PSCI`).

Not all timestamps are captured in all cases.  However, whenever timestamps are captured, they do so in pairs.

vingu-linaro, 2017-06-01:
Thanks for the additional details. That's what i have started to do and you will save me time to figure out if my tid was correct

vingu-linaro, 2017-06-07:
Thanks for your help.
I have enable rt instrumentation for hikey platform and will send a pull request soon. I have also a patch to get instrumentation figures from linux kernel that i'd like to send on lkml for review as well.

dp-arm, 2017-06-07:
@vingu-linaro Can you keep CC us for the kernel patches as well?

vingu-linaro, 2017-06-07:
Yes I will keep you CC

danh-arm, 2017-06-19:
This issue seems to be resolved

closed 2017-06-19 by danh-arm