ARM-software/tf-issues #487: mbedtls_common.c doesn't compile if MBEDTLS_PLATFORM_SNPRINTF_ALT is not defined

sbranden, 2017-06-03:
mbedtls_common.c doesn't compile if MBEDTLS_PLATFORM_SNPRINTF_ALT is not defined.

This is because mbedtls_platform_set_sn_printf is function prototype is only defined if MBEDTLS_PLATFORM_SNPRINTF_ALT is defined.

danh-arm, 2017-06-05:
That is intentional. We expect MBEDTLS_PLATFORM_SNPRINTF_ALT to always be defined, and the provided tf_snprintf to always be used, since this is smaller than (and the same speed as) typical snprintf implementations. Do you need to use your own snprintf implementation for some reason?

antonio-nino-diaz-arm, 2017-06-05:
By the way, how are you compiling the mbedtls module? As far as I see, this problem can't happen if you use the provided C/H files and makefiles. For `MBEDTLS_PLATFORM_SNPRINTF_ALT` to not be defined you need to either remove the define from `mbedtls_config.h` or change the paths in `mbedtls_common.mk` and point at your own copy of the configuration file (or have your own makefile for the mbedtls module pointing at your files). In the first case, if you are modifying the module in your copy of the Trusted Firmware, you might as well remove the problematic line in `mbedtls_common.c`. In the other case, you can just copy `mbedtls_common.c` and remove the line in your copy.

sbranden, 2017-06-05:
Yes, we compile with our own mbedtls_config.h.

For us, it's not really a problem as I added the define for MBEDTLS_PLATFORM_SNPRINTF_ALT.

Your comment about just making a copy of mbedtls_common.c doesn't sound like a good solution though.  It would be much better to just add an #ifdef MBEDTLS_PLATFORM_SNPRINTF_ALT around the call instead.

danh-arm, 2017-06-06:
In general, the upstream code does not support platforms having their own mbedtls_config.h. This is generic configuration. However, in this specific case it's easy to support a configurable `MBEDTLS_PLATFORM_SNPRINTF_ALT` setting by adding the `#ifdef` you suggest. We will create a patch to this effect.

antonio-nino-diaz-arm, 2017-06-12:
I forgot to add the reference of this issue to the commit in https://github.com/ARM-software/arm-trusted-firmware/pull/968 and that's why the issue hasn't been closed automatically.

closed 2017-06-12 by antonio-nino-diaz-arm